epics-modules/opcua #7: Need to support blanks in Node definition

bkuner, 2018-11-07:
As we mailed, there is a need of support blanks in node - because it is an allowed character for a node definition - as well as the cruel ". We can ask our own programmers to avoid this but not externals.
My suggestion is to implement the link as an name=value; list, maybe with user definable seperator for the node and more parameters. 
E.g. 'SUBSCRIPTION_TAG ns=2;s=what. is this; monitor=y,somthing=else'

bkuner, 2018-11-07:
Sorry, typo:
E.g. 'SUBSCRIPTION_TAG ns=2;s=what. is this; monitor=y;somthing=else'

bkuner, 2018-11-07:
It is your program, so do as you want. To me all quoted characters are bad and uggly and I would avoid it if possible.( I don't see how to avoid quoted double quotes)
What you do is you actually define a list of data in the link and use the space as delimiter. This is of very common use in unix environmen and gnumake but not in windows environment and the programming languages I know. To me it looks better without quotes and a defined delimiter (that is not a space).
Seems space delimiters are for old people like us that complain about modern things as spaces in filenames, spaces in identifiers...

ralphlange, 2018-11-07:
The OPC spec allows any series of (up to 4096) non-null UTF-8 characters.
That is way more flexibility than I could explain with age. While spaces certainly seem reasonable, I do not see the need to support emoji or non-printable characters.

Double quotes will have to be escaped, correct - that's an EPICS db thing.

ralphlange, 2018-11-07:
The only safe way seems to be putting the identifier to the end of the value - then it is always null terminated, matching the OPC UA spec.

ralphlange, 2018-11-07:
BTW. Double quotes as part of the name are the default for Siemens TIA Portal (i.e. the S7-1500 server) - I think we should support that.

bkuner, 2018-11-08:
We don't need to discuss about emojis, my point is to think about a reasonable delimiter for a list. To me the space is not a good choice because spaces come in users mind more and more in use as an ordinary character - for filenames, identifiers etc as I mentioned before. 
I would prefere any solution that relates to widley used programming languages - e.g. the semicolon (csv format, c, cpp, perl...) or any good parsable syntax for lists and also dictionaries - because that is what you have in the link and that is what users have to write to the link. Spaces for that purpose are in use only in odd things like shell, -scripts and makefiles!
My last comment to this - it is your program.

ralphlange, 2018-11-08:
Which programming language was it that allows spaces in variable names?

bkuner, 2018-11-08:
Whitespace :-)
Ha, by the way our non printable characters could also be some stuff that is quite normal text for asian people etc. And why not define links with emojies?? - Not my way, but there are many ways :-)

bkuner, 2018-11-08:
opcua - as well

ralphlange, 2018-11-08:
All fine and well justified.
I just wrote that I think emojis or non-printable characters would be too much.

ralphlange, 2018-11-08:
Will find a way to support everything else. Including semicolons.

bkuner, 2018-11-08:
Isn't it done if we define OUR syntax for list and dict and to give the best support for our users if we do this that most of them can imagine how it works without studying to much docu?
The usual way is to define a rarely used character and to quote it if it has another meaning - and to me this is not the space.

bkuner, 2018-11-08:
Do you remember our epics xls syntax, there it was your idea to use the '|' as delimiter - rare enough. Now as we can write ordinary records to it - e.g. calc we also run into the problem to quote this in CALC expressions. You can't avoid this problem - you only can make it the way that you don't run into every day.

ralphlange, 2018-11-08:
So a configurable separator plus escaping would be good enough, right?

Question: The `;` is part of the standard after `ns=<n>` - should it always be a separator, or only at this place?

bkuner, 2018-11-08:
Would be my choice - for this reason. Also NAME=VALUE for other options.
We could think about double quoting strings (with quoted double quotes inside) to make spaces optional - but increase parse complexity OR prohibit extra spaces.

bkuner, 2018-11-08:
Consecuently  use of the semicolon would be:
SUB;ns=2;s=\"what ever\; or not\";monitored=y
or 
SUB;ns="2";s="\"what ever\; or not\"";monitored="y"
I would prefer the first

bkuner, 2018-11-08:
Ups the system eat my quotes for the quoted quotes

ralphlange, 2018-11-08:
Note that these quotes would have to be escaped in the db file, and that quotes are the default in TIA Portal as part of the name. The first case becomes something like:
```
field (INP, "@SUB;ns=2;s=\\\"\"Data_block_1\".\"my Variable\"\\\";monitored=n")
```
The second case I won't even think of.

bkuner, 2018-11-08:
Ok. I would like this. Unfortunately we need the many backlashes for the EPICS

ralphlange, 2018-11-08:
For someone who said escaping is ugly, you are making a strong statement... ;-)

ralphlange, 2018-11-08:
My simplistic approach (escaping the space) would look like:
```
field (INP, "@SUB ns=2;s=\"Data_block_1\".\"my\\ Variable\" monitored=n")
```

ralphlange, 2018-11-08:
Having a configurable seperator plus escaping would allow both.

bkuner, 2018-11-08:
Sorry, first think about. This is my:
`field (INP, "@SUB;ns=2;s=\"Data_block_1\".\"my Variable\";monitored=n")`

ralphlange, 2018-11-08:
Yes, and in that case separator plus escaping would allow both ways.
Sold!

bkuner, 2018-11-08:
Aren't you EPICS core developer? What about definig a more flexible input links to avoid all escape chars? - e.g. definition of not interpreted string as in perl, python with single quotes - that is very helpfull there. Remember perl comand line one-liners:
`field (INP, '@SUB;ns=2;s="Data_block_1"."my Variable";monitored=n')`

bkuner, 2018-11-08:
At the epics-controls.org up to now I found only what I allready know - This getting started stuff. I didn't pick the raisin "What features does it bring" that helps me to learn the new features in the server, the channel access kompared with our 3.14 and how to pv-access. May be it is there - but I didn't get it.

closed 2018-11-14 by ralphlange

ralphlange, 2018-11-14:
I implemented a minimal simple solution:
As before: __Space and semicolon are separators.__
Now: __Any separator can be escaped by preceding it with two backslashes.__
(Needs to be two because the first backslash is eaten by the database parser.)

This should allow you to go on with testing.
If you still think making the separator configurable is a good idea, please open a new issue that references this one (i.e. something like "see #7" is good enough as description).

commit e73acbdf (2018-11-14): linkParser: allow escaping separators (fixes #7)