epics-modules/opcua #8: Allow DESC field to be set from OPC UA item

ralphlange, 2018-11-07:
**OPC UA items have a description that could be used to set DESC**
There is an information structure for all OPC UA items that contains a string field with a description. This could be read at connection time and used to set the DESC field of the record that connects to the item.

Since this description might not contain useful information on a specific OPC UA server, and to avoid clashing with DESC settings in the EPICS database, this should be configurable per record (option) with a global default (variable) that can be set from the startup script.

tim-speight, 2025-01-17:
Not a runtime facet. But the generic XSLT I shared with you last month might be a good way to bring a "desc" directly to EPICS Db records.  I'm still refactoring this now for specific purpose, things like flippers and having an _In/_Out suffix based as a naming convention.

ralphlange, 2025-01-17:
Good point!

ralphlange, 2025-01-17:
@dirk-zimoch 
That XSLT takes the XML description of a server (that it produces according to OPC UA spec) to create an EPICS database.

Of course, that is rarely what you want in the end, but it gets you generic full access to everything on the server in an instant.