epics-modules/opcua #9: exampleTop/opcuaIocApp won't compile here

bkuner, 2018-11-13:
without adding libcrypto to opcuaIoc libs:
Makefile:
```
# Add all the support libraries needed by this IOC
opcuaIoc_LIBS += opcua
opcuaIoc_SYS_LIBS += crypto

```

ralphlange, 2018-11-13:
Example in place or copied out?
Which version of EPICS Base?
What's your setting of UASDK_DEPLOY_MODE?

Please try to supply enough information that I can try to reproduce the behavior.

ralphlange, 2018-11-13:
What is your setting for UASDK_USE_CRYPTO in CONFIG_SITE(.local)?

ralphlange, 2018-11-13:
My linker line looks exactly the same, but I am using SDK 1.5.5, and the binary does not need to be linked against libcrypto.

How did you build the SDK? Did you use the standard script (that actually does not build all libraries as shared libraries even if you tell it to)? I am patching the Unified Automation build script to do clean shared library and static library builds.
I think if you use the original build script, the one static library contains calls to libcrypto symbols creating a dependency even if you don't use the crypto things in the SDK.

ralphlange, 2018-11-13:
Nevertheless: if `UASDK_USE_CRYPTO = YES`, libcrypto should appear on the linker line, so I will consider this a bug.

bkuner, 2018-11-13:
don't use any local files here
```
# How the Unified Automation SDK shared libraries are deployed
#   SYSTEM = shared libs are in a system location
#   PROVIDED = shared libs are in $(UASDK_DIR)
#   INSTALL = shared libs are installed (copied) into this module
UASDK_DEPLOY_MODE = PROVIDED
UASDK_DIR = $(UASDK)/lib
# How the Unified Automation SDK libraries were built
UASDK_USE_CRYPTO = YES
UASDK_USE_XMLPARSER = YES

USR_CXXFLAGS_Linux += -std=c++11
USR_CXXFLAGS += -DUSE_TYPED_RSET

```

commit 2e54a8c4 (2018-11-13): uasdk/rules: link libcrypto if UASDK_USE_CRYPTO=YES (fixes #9)
closed 2018-11-13 by ralphlange