epics-modules/opcua #36: Max-items-per-service-call does not work with UA SDK

ralphlange, 2019-06-19:
**Bug Description**
The UA SDK feature of setting the maximum number of items per service call is buggy. The setting is only honored by _synchronous_ OPC UA service calls and does not work for the _asynchronous_ call variants.

**To Reproduce**
Steps to reproduce the behavior:
1.  Setup an S7-1500 server (that has a limit of 1000 items per service call)
2.  Create IOC database that puts >1000 items on a single subscription
4.  The subscription will not work at all

**Setup**
-   OPCUA Support: 0.4.0
-   Platform: RHEL7.4
-   EPICS Base: 7.0.2
-   Client library: UA SDK 1.5.5
-   Server: S7-1500

closed 2019-07-08 by ralphlange

jeonghanlee, 2019-10-31:
@ralphlange, is this really fixed? With 0.5.2, try to connect the simens PLC. The nodes more than 1000, readComplete returns "zero length" values at https://github.com/ralphlange/opcua/blob/master/devOpcuaSup/UaSdk/SessionUaSdk.cpp#L560
 After this, I've got the following message, and the ioc was crashed. 

````
############################################################################
OPC UA Client Device Support v0.5.2; using Unified Automation C++ Based OPC UA Client SDK
OPC UA: Autoconnecting sessions
OPC UA session OPCUA-MPSID: connection status changed from Disconnected to Connected
OPC UA session OPCUA-MPSID: (readBatchOfNodes) beginRead service ok (transaction id 1; retrieving 1112 nodes)
Session OPCUA-MPSID: (readComplete) getting data for read service (transaction id 1; data for 0 items)
```

The workaround which I did is to force to set the batch-nodes 1000 such as
```
opcuaSetOption $(SESSION) batch-nodes 1000
```
Then the ioc returns 

```
OPC UA session OPCUA-MPSID: connection status changed from Disconnected to Connected
OPC UA session OPCUA-MPSID: (readBatchOfNodes) beginRead service ok (transaction id 1; retrieving 1000 nodes)
OPC UA session OPCUA-MPSID: (readBatchOfNodes) beginRead service ok (transaction id 2; retrieving 112 nodes)
Session OPCUA-MPSID: (readComplete) getting data for read service (transaction id 2; data for 112 items)
Session OPCUA-MPSID: (readComplete) getting data for read service (transaction id 1; data for 1000 items)
```

Can I create an issue instead of this? What do you prefer?

ralphlange, 2019-11-04:
A crash is always a bug - please create an issue for this. Looks like the unsuccessful read service return (`readComplete()` with data for 0 items) is not detected and/or handled correctly.

However, the rest of your description seems like the feature is working as expected.
If you don't set the batch size, the `read()` tries to do all 1112 nodes in one request, which does not work against your 1500.
If you set the batch size option, the `read()` correctly creates two requests: one with the max. batch size (1000), one with the remaining items (112). Both read requests return successfully.

What behavior did you expect?
(Note that the 1000 items per call limitation is only on the server side in this specific server implementation, and not advertised other than in the Siemens documentation.)

jeonghanlee, 2019-11-04:
I have to talk a system owner in order to collect the crash information. Once I do, I will create an issue for this.  

Yes, I liked your implementation. In case, I added that limitation as a default configuration https://github.com/icshwi/e3-opcua/blob/master/iocsh/opcua.iocsh#L24 for people who doesn't aware about them in order to protect any IOC passively

ralphlange, 2019-11-04:
Good.
(I was afraid you were asking for something "automatic", which ... at least would be really ugly no matter how. Putting knowledge about a specific server into the generic client... uargh.)

f4c09b35d26e219d5f71bf51ad7dad09dfabadbf adds a paragraph about this limit (and how to get around it) in the S7-1500 example README.

commit 9e8e3521 (2019-07-08): uasdk: fix broken batching for asynchronous SDK calls

(fixes #36)
commit 2cb41dcf (2020-05-26): uasdk: fix broken batching for asynchronous SDK calls

(fixes #36)