epics-modules/opcua #61: Add connection status PV (by session)

ralphlange, 2020-07-27:
SSIA.

There needs to be support for an mbbi "connection status" PV per session, so that connection history and status can be seen and archived without going to the console log.

ralphlange, 2020-09-15:
Pushed to 0.8, as 0.7 is delayed.

ralphlange, 2021-07-05:
Too invasive to be introduced at this point. Deferring.

tim-speight, 2025-02-10:
I Up-vote this development effort having this metric would help to see if a PLC has fallen over or some TCP networking event occured.

OPCUA has a serviceConnection context and there are items in at the 'session level' > https://documentation.unified-automation.com/uasdkcpp/1.8.3/html/classSession.html

It would be good if we can have readOnly status for variables like HOSTNAME, TIMEOUT, hasExpired, isActivated + isValid + isSecureChannelValid.  

I don't know much about OPCUA but for keeping a TCP link alive; it might be useful to also expose a writeable to resetTimeout to allow for a type of ping for any long running TCP connects and against any long running EPICS ioc.

A nice to have if it is all on the secureChannels layer within OPCUA - knowing the userName could prove to be useful?