epics-modules/opcua #63: The latest SDK evaluation version question

jeonghanlee, 2020-08-12:
**Describe the bug**
Compiling failure

**To Reproduce**
Steps to reproduce the behavior:
1.  Extract the latest SDK into `/home/jhlee/programs/UASDK`

2. Set the following variables in `configure/CONFIG_SITE.local`

```bash
UASDK:=/home/jhlee/programs/UASDK
STATIC_BUILD=YES
SHARED_LIBRARIES=NO
```

3. `make distclean` and `make` 

```bash
/usr/bin/g++  -D_GNU_SOURCE -D_DEFAULT_SOURCE            -D_X86_64_  -DUNIX  -Dlinux      -O3   -Wall     -std=c++11  -mtune=generic      -m64  -I. -I../O.Common -I. -I. -I.. -I../../include/compiler/gcc -I../../include/os/Linux -I../../include   -I../../../include -I/home/jhlee/epics-7.0.4/epics-base/include/compiler/gcc -I/home/jhlee/epics-7.0.4/epics-base/include/os/Linux -I/home/jhlee/epics-7.0.4/epics-base/include        -I/home/jhlee/programs/UASDK/include -I/home/jhlee/programs/UASDK/include/uaclientcpp -I/home/jhlee/programs/UASDK/include/uapkicpp -I/home/jhlee/programs/UASDK/include/uabasecpp -I/home/jhlee/programs/UASDK/include/uastack -I/home/jhlee/programs/UASDK/include/xmlparsercpp    -c ../opcuaIocMain.cpp
/usr/bin/g++ -o opcuaIoc -Wl,-Bstatic -L/home/jhlee/gitsrc/EPICS-env/opcua-src/lib/linux-x86_64 -L/home/jhlee/gitsrc/EPICS-env/opcua-src/exampleTop/lib/linux-x86_64 -L/home/jhlee/epics-7.0.4/epics-base/lib/linux-x86_64 -L/home/jhlee/programs/UASDK/lib -Wl,-rpath,/home/jhlee/gitsrc/EPICS-env/opcua-src/lib/linux-x86_64 -Wl,-rpath,/home/jhlee/gitsrc/EPICS-env/opcua-src/exampleTop/lib/linux-x86_64 -Wl,-rpath,/home/jhlee/epics-7.0.4/epics-base/lib/linux-x86_64 -Wl,-rpath,/home/jhlee/programs/UASDK/lib     -Wl,--disable-new-dtags       -rdynamic -m64         opcuaIoc_registerRecordDeviceDriver.o opcuaIocMain.o    -lopcua -ldbRecStd -ldbCore -lca -lCom -luaclientcpp -luapkicpp -luabasecpp -luastack -lxmlparsercpp -Wl,-Bdynamic -lxml2 -lcrypto -lpthread   -lreadline -lm -lrt -ldl -lgcc
/usr/bin/ld: cannot find -luastack
collect2: error: ld returned 1 exit status
 ```
because the latest SDK have all static libraries and has only libuastack.so as dynamic.  
 
4. So, I went to into `exampleTop/opcuaIocApp/O.linux-x86_64`, then change libuastack to be in dynamic area such as

```bash
O.linux-x86_64 (master)$ /usr/bin/g++ -o opcuaIoc -Wl,-Bstatic -L/home/jhlee/gitsrc/EPICS-env/opcua-src/lib/linux-x86_64 -L/home/jhlee/gitsrc/EPICS-env/opcua-src/exampleTop/lib/linux-x86_64 -L/home/jhlee/epics-7.0.4/epics-base/lib/linux-x86_64 -L/home/jhlee/programs/UASDK/lib -Wl,-rpath,/home/jhlee/gitsrc/EPICS-env/opcua-src/lib/linux-x86_64 -Wl,-rpath,/home/jhlee/gitsrc/EPICS-env/opcua-src/exampleTop/lib/linux-x86_64 -Wl,-rpath,/home/jhlee/epics-7.0.4/epics-base/lib/linux-x86_64 -Wl,-rpath,/home/jhlee/programs/UASDK/lib     -Wl,--disable-new-dtags       -rdynamic -m64         opcuaIoc_registerRecordDeviceDriver.o opcuaIocMain.o    -lopcua -ldbRecStd -ldbCore -lca -lCom -luaclientcpp -luapkicpp -luabasecpp  -lxmlparsercpp -Wl,-Bdynamic -lxml2 -lcrypto -lpthread   -lreadline -lm -lrt -ldl -lgcc -luastack
```

5. I have `opcuaIoc` in that directory.

**Expected behavior**

**Setup (please complete the following information if applicable):**
-   OPCUA Support: master branch

-   Platform: 

```bash
Linux 4.19.0-10-amd64 #1 SMP Debian 4.19.132-1 (2020-07-24) x86_64 GNU/Linux
gcc (Debian 8.3.0-6) 8.3.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

-   EPICS Base: 7.0.4
 
-   Client library: uasdkcppbundlepubsub-bin-EVAL-debian9.2.3-x86_64-gcc6.3.0-v1.7.2-493.tar.gz

-   Server: N/A

**Additional context**

I am coming back to OPCUA world now.

ralphlange, 2020-08-12:
The EVAL distribution from Unified Automation does not support static builds.
I assume that is done intentionally to avoid people creating static binaries from EVAL. They keep the build dynamic, so that any binary will need `libuastack.so` which you have signed to not distribute. (See evaluation license.)

If you leave `STATIC_BUILD` as `NO` (default), building against the EVAL works as expected, using dynamic libraries if possible and static libraries if there are no shared libraries available.
Why do you want to build statically, anyway?

Your point of using `INSTALL_LOCATION` for the DBD include path is correct. I will add the fix.

ralphlange, 2020-08-12:
The example is an "embedded TOP" structure, with a configuration of its own.

For the "embedded" use case of the example, `OPCUA` should be set to "the `INSTALL_LOCATION` setting of the parent module". This is something not easily achievable in the `RELEASE` file, which gets parsed by a Perl script, so any sophisticated Make commands are not available.

I will add text to the README files for both issues.

closed 2020-08-12 by ralphlange

jeonghanlee, 2020-08-13:
Thank you, @ralphlange.  I am waiting for the purchase the license. That is the reason why I use the evaluation version. Stay well and be safe!

ralphlange, 2020-08-13:
np.
Why were you trying to build statically? Any specific reason, or just "because I can"?
Also: check the links to the static IOC that I sent around. There is an ubuntu version.

jeonghanlee, 2020-08-13:
I couldn't compile it, because the latest evaluation version does have all static libraries except libuastack{d}.so. I have no choice. I was frustrated, because I deleted all evaluation and license versions at the day of license transfer in my previous company. 
With the final manual intervention, I have the final opcusIoc binary with mixed version (static and dynamic). At least I can see the ioc console. I will test it later.

ralphlange, 2020-08-13:
But a regular (shared) build "just works" [TM] against the EVAL, taking the stack as shared and the others as static.
The Travis builds are all done using EVAL bundles, without any special setting.

ralphlange, 2020-08-13:
You trying to set the build to STATIC = YES was making it fail and creating your issues.

(Seems you found yet another elegant way to shoot yourself in the foot...)

jeonghanlee, 2020-08-13:
I am a person who has always the bad luck. That makes my life a bit miserable. You are right, I don't want to do, but seems no way to escape it. 
I didn't use `STATIC=YES`, guess why I was there.   Maybe I need just you to talk :)

Anyway thanks, @ralphlange. Much appreciate.

ralphlange, 2020-08-13:
Born beta tester.

commit e5ff4ecd (2020-08-12): uasdk: Add note about use of the evaluation binary bundle

(solves remaining part and fixes #63)