epics-modules/opcua #77: Usage of opensource implementation for OPC-UA

jalalmostafa, 2020-11-20:
The UA SDK is quite nice, but there are many other open-source alternatives e.g. https://open62541.org/

With open-source, developers can get unlimited access to updates, maintenance, and support.

ralphlange, 2020-11-20:
Yes.
Given that each client library / SDK has its own API and they are quite different, that would need a considerable amount of work.
On my part, the Device Support architecture is specifically done in a way that will allow to add other client libraries. Using open source was the plan from the very beginning of the project. At that time, none of those alternatives had the levels of feature-completeness and quality necessary for production systems. Meanwhile that has changed, and open62541 would be the primary  candidate in my opinion, for sure.

Do you consider to contribute by adding an open-source alternative, based on e.g. open62541?

ralphlange, 2021-01-18:
_Update:_ @dirk-zimoch and Carsten Winkler (HZB/BESSY) have started on this journey.

ralphlange, 2021-01-25:
Branch [feature/open62541-integration](https://github.com/ralphlange/opcua/tree/feature/open62541-integration) is dedicated to this project.

ralphlange, 2021-01-26:
I created the `feature/open62541-integration` branch in this repo (upstream fork) as a target for all Pull Requests within the scope of this activity.

Reason: The "normal" development of the Device Support will continue, this will not be a short activity, and I would like this addition to be somewhat robust and stable before merging into master. Merging now (and often) - at this point in the activity - would be likely to interfere with regular development.

So: please develop the open62541 integration in your personal forks - separate things could/should be done in separate branches.
Once you want more people to have a look at what you did or you think it is ready, create a Pull Request into `feature/open62541-integration` of the upstream fork. Once things are discussed and deemed fine, we'll merge it there.

Only when the whole combined effort of the open62541 integration (represented by the `feature/open62541-integration` branch of the upstream fork) is in a working and robust state, we should discuss to merge it into `master` to be released.

ralphlange, 2021-08-05:
Scheduling integration of this effort as an experimental feature for 0.10.
Should be limited to basic type variables at this point.

ralphlange, 2023-08-02:
Closed with cc8f153ee83b7efe70e904130db0a9a0984ab211.

closed 2023-08-02 by ralphlange