epics-modules/opcua #90: problem starting example IOC iocopcuaEx

dirk-zimoch, 2021-02-01:
When starting the IOC (using EPICS 7.0 on Linux), it hangs forever (?) on the st.cmd line
`dbLoadDatabase("../../dbd/opcuaIoc.dbd",0,0)`

The arguments 0,0 seem to be for VxWorks only, not for Linux. No idea why it actually hangs (have not investigated)
Changing the line to this fixes it:
`dbLoadDatabase("../../dbd/opcuaIoc.dbd")`

ralphlange, 2021-02-01:
These additional arguments are in the `st.cmd@Cross` version of the startup command file in [3.15](https://github.com/epics-base/epics-base/tree/3.15/src/template/base/top/iocBoot/ioc) and [7.0](https://github.com/epics-base/epics-base/blob/7.0/modules/database/src/template/top/iocBoot/ioc/st.cmd%40Cross).

They are in none of the other (Common, RTEMS, vxWorks) versions.

Works fine on my Linux systems. (Debian 9, Debian Testing, RHEL 7)
Will fix anyway.

ralphlange, 2021-02-01:
```C
/* dbLoadDatabase */
static const iocshArg dbLoadDatabaseArg0 = { "file name",iocshArgString};
static const iocshArg dbLoadDatabaseArg1 = { "path",iocshArgString};
static const iocshArg dbLoadDatabaseArg2 = { "substitutions",iocshArgString};
static const iocshArg * const dbLoadDatabaseArgs[3] =
{
    &dbLoadDatabaseArg0,&dbLoadDatabaseArg1,&dbLoadDatabaseArg2
};
static const iocshFuncDef dbLoadDatabaseFuncDef =
    {"dbLoadDatabase",3,dbLoadDatabaseArgs};
static void dbLoadDatabaseCallFunc(const iocshArgBuf *args)
{
    iocshSetError(dbLoadDatabase(args[0].sval,args[1].sval,args[2].sval));
}
```

commit adb784ee (2021-03-03): example: use Linux version of st.cmd

- fixes #90
closed 2021-03-03 by ralphlange