epics-modules/opcua #95: Write errors should set SEVR/STAT to INVALID/WRITE

dirk-zimoch, 2021-02-18:
Writing an array (aao record) with a wrong length fails and reprocesses the record with reason=writeFailure. However, this does not set the record to INVALID/WRITE alarm state.

Other record types (and other error conditions) may be affected as well.

dirk-zimoch, 2021-02-18:
Tested with open62541implementation. I don't know how the UA_SDK implementation handles arrays with mismatching length.

dirk-zimoch, 2021-02-19:
When initial read fails, the records are INVALID/UDF instead of INVALID/READ. Input records stay like this even when actively read again, output records become NO_ALARM (!) without doing anything.
I have experienced this with the server error "License expired".  :-)

Pervious versions of this drivers showed a different bahavior. The records would go to INVALID/COMM alarm because of an exception thrown ("no incoming data").

ralphlange, 2021-02-19:
The error handling was added (some redone) to improve the "no incoming data" INVALID/COMM behavior.
Obviously an incomplete project.

I agree with your suggestion of INVALID/READ after a failed initial read. Also for output records? Probably.

dirk-zimoch, 2021-02-19:
The "improvement" was unfortunately degrading. I do not care too much if the error is COMM or READ,WRITE. The latter is more specific: initial read error of an output record deserves a READ alarm (one could agrue for UDF) while failing write should get a WRITE alarm, but both are undoubtly communication errors and may as well be COMM alarms. I do not favor UDF as that is typical for records that have never been processed. And NO_ALARM after a write failure is just wrong.

ralphlange, 2021-02-19:
The old behavior was ignoring all errors. That led to an exception being thrown during record processing that was caught and put as INVALID/COMM.
Hard to degrade from that, but possible.

ralphlange, 2021-02-19:
I will leave INVALID/COMM for communication failures that are not directly related to a read or write operation. (Disconnect etc.)