epics-modules/opcua #102: Client does not write until reboot, if it gets "BadInvalid" state from the server at init

moratom, 2021-04-13:
**Description**
We have a client that only writes to the server, and we do not want it to do anything at init.
```
record(ao, "$(P)$(R)_Centroid_X"){
    field(DTYP, "OPCUA")
    field(SCAN, "1 second")
    field(DOL, "$(SCREEN)Stats:CentroidX_RBV")
    field(OIF, "Full")
    field(OMSL, "closed_loop")
    field(OUT, "@EPICS ns=2;s=$(OPC_P)$(R)X_mon monitor=n bini=ignore")
}
```
However, if the state of the node on the OPC UA server is `BadInvalidState`, it doesn't write the data but stays in the state `initialRead` and it doesn't recover when the state of the node goes to `Good` and never writes until the restart of the IOC.
```
epics> opcuaShowData OPCUA:Screen1_Centroid_X
item ns=2;s=exnBTF_out_Screen1X_mon record=OPCUA:Screen1_Centroid_X state=initialRead status=BadInvalidState context=@EPICS sampling=0(-1) qsize=0(1) cqsize=3 discard=old timestamp=server bini=ignore output=y monitor=n registered=-(n) leaf=[ROOT] record(ao)=OPCUA:Screen1_Centroid_X type=OpcUa_Double timestamp=server bini=ignore monitor=n
```

**Expected behavior**
The IOC would consistently try to write to the OPC UA server, or detect when the state is not `BadInvalid` anymore.

**Setup:**
-   OPCUA Support: 0.8.0
-   Platform: Linux Kubuntu 20.04 LTE
-   EPICS Base: R7.0.4.1
-   Client library: UA SDK v1.7.4-520

ralphlange, 2021-04-13:
Well, the intention is that the `bini` setting controls the behaviour when the _session_ comes up. I actually never thought about a specific item being in invalid state.

As `bini=ignore` configures the IOC to _not_ write when the connection comes up, but only on record processing...
Are you saying that such explicit writes (e.g. by CA put operations) are not working until the IOC is rebooted?

Can you check if things get back to normal when you disconnect and reconnect the session? Does the record stay in the `intialRead` across a disconnecting session? That would clearly be a misbehaviour.

I am not a fan of the IOC checking for the state of invalid nodes. That's not really feasible in a scalable way.
But the `BadInvalidState` return should definitely get that record out of the `initialRead` state into something that makes it retry a write when the record processes. Sounds like something that I did not test for.

moratom, 2021-04-13:
> Are you saying that such explicit writes (e.g. by CA put operations) are not working until the IOC is rebooted?
>
> Can you check if things get back to normal when you disconnect and reconnect the [...]

It does work, if you disconnect and reconnect to the session, but doesn't work by explicit writes until you do that.

> But the `BadInvalidState` return should definitely get that record out of the `initialRead` state into something that makes it retry a write when the record processes. Sounds like something that I did not test for.

That would solve it for our use case yes.

ralphlange, 2021-04-13:
Ok, I will look into that. Sorry.
Will ping you when I have something to test.

ralphlange, 2021-04-14:
Here's a patch against release 0.8.0 that I would like you to test.

I don't have a server with an item that shows the behaviour you describe, so I need your support to verify the fix. I tried to verify as good as possible (manually) that the behaviour against "normal" items is as before.

[0001-Process-and-unlock-record-if-initialRead-fails.patch.txt](https://github.com/ralphlange/opcua/files/6311368/0001-Process-and-unlock-record-if-initialRead-fails.patch.txt)

moratom, 2021-04-16:
I tested it today and the patch works as expected. Thank you!

ralphlange, 2021-04-16:
Excellent. Thanks!

I will merge so that this becomes part of v0.9
Do you need a tagged released bugfix v0.8.1 or are you fine working with the patch until 0.9 is released (~few weeks)?

moratom, 2021-04-19:
I'm fine working with the path. Thank you again for the fix.

ralphlange, 2021-04-19:
Welcome!

commit 38feb0d9 (2021-04-19): Process and unlock record if initialRead fails

- fixes #102
closed 2021-04-19 by ralphlange