epics-modules/opcua #104: should not install rules

bfrk, 2021-04-15:
Using the latest released version v.0.5.2.

`devOpcuaSup/UaSdk/Makefile` installs build rules:

```
CFG += RULES_OPCUA
CFG += CONFIG_OPCUA
```

This is bad and should not be needed. It is bad, because e.g. the following line in RULES_OPCUA

USR_SYS_LIBS += $(UASDK_LIBS)

means that _every_ program or library built inside a module or application that has opcua support in its configure/RELEASE will now be linked against the UASDK_LIBS.

It is unneeded because I have removed the make rule installation and can build our applications just fine w/o them.

ralphlange, 2021-04-15:
The latest released version is v0.8.0.

I get your point. It is true that some types of builds (including yours) might not need the rules. Others do.

bfrk, 2021-04-15:
> The latest released version is v0.8.0.

Sorry.

> I get your point. It is true that some types of builds (including yours) might not need the rules. Others do.

I wouldn't mind the extra rules if they didn't break our builds. The problem is that these libraries don't exist for all target archs (e.g. RTEMS and other cross archs). Generally, I don't think it is correct to set things like USR_LIBS in rules published by a support module as that forces all downstream static libraries and binaries to link them. I think it would be better to just define a variable that contains all the needed libraries and let users do their own xxx_LIBS +=, just as they have to do for any other libraries (base or support modules).

bfrk, 2021-04-15:
BTW, the actual rules (not variable definitions) in RULES_OPCUA look like they are for use within opcua itself, not for users to include.

ralphlange, 2021-04-15:
Depends on the build type (SYSTEM | PROVIDED | EMBED).

ralphlange, 2021-04-15:
But your point about not setting the global variables is valid.

closed 2021-05-23 by ralphlange

ralphlange, 2021-05-23:
As discussed:
I left the four deploy types around, as they all seem to have valid use cases. The remaining parts of the PR (#106) are merged with a few changes.
Thanks for pointing this out!

commit 5a70d736 (2021-05-23): test: fix linking by using the new variables

- closes #104, closes #106.
commit 85d093b1 (2021-05-23): Make build system settings less invasive (fix #104)

This makes the module less invasive w.r.t how user applications are built.
We no longer install RULES_OPCUA; this is only used internally now. Users
now have to add suitable definitions to their Makefiles, as exemplified in
the exampleTop/opcuaIocApp/Makefile:

opcuaIoc_LIBS += opcua $(UASDK_LIBS)
opcuaIoc_SYS_LIBS_Linux += $(UASDK_SYS_LIBS_Linux)

The installed CONFIG_OPCUA defines the UASDK_LIBS and UASDK_SYS_LIBS_Linux
variables according to the chosen configuration.