epics-modules/opcua #110: Waveform Records not getting Time from Server

JTGWilson, 2021-05-11:
**Description**
Waveform records are not using the Timestamp from the OPC UA server, even when the field TSE = -2.

I have an OPC UA Server and Client setup in LabVIEW. The Server is hosting several Nodes which are arrays of Long data. 
A LabVIEW OPC UA Client reads the data and Timestamps as set by the Server. 
Replicating the same in EPICS results in the Data being read correctly, but the Timestamps are not what is being set by the OPC UA Server.  

The same setup in EPICS but with the Nodes changed to scalar Long results in the EPICS PVs having the OPC UA Server Timestamps as expected. 

**To Reproduce**
Steps to reproduce the behavior:
1.  Setup OPC UA Server to generate arrays of data (in my case 7500 elements in each). 
2.  Create IOC database that has Waveform records which will be connected as OPC UA Clients and set field(TSE, "-2")
3.  Start IOC with the normal opcuaCreateSession etc. 
4.  camonitor of the Waveform PVs shows that the Data is coming in as expected but the Timestamps don't match up what is being set by the OPC Server.

**Expected behaviour**
The Waveform PVs would have the OPC UA Server Timestamps as their Timestamps if you specify that the field TSE is -2.

**Setup:**
-   OPCUA Support: v0.8.0
-   Platform: CentOS 8
-   EPICS Base: base-7.0.3.1
-   Client library: UA SDK EVALUATION v1.7.4-520
-   Server: LabVIEW with OPC UA Toolkit

ralphlange, 2021-05-11:
Very strange, indeed.

I just checked the code for the array and the scalar read, and the EPICS time stamp is being set in the same way and the same place in both cases. Same for the time stamp and TSE "-2" handling in EPICS Base itself.
Need to set up a test trying to see this behavior.

ralphlange, 2021-05-13:
I have set up a demonstration using the `UaServerCpp` demonstration server that is part of the UA bundle.

The IOC opens a single session with the demo server and adds a subscription with 200ms publishing interval. It loads a waveform record for the array of int32 variable that the demo server offers:
```
record(waveform, "$(P)$(R)wfint32") {
    field(DTYP, "OPCUA")
    field(INP,  "@$(SUBS) ns=2;s=Demo.Dynamic.Arrays.Int32")
    field(FTVL, "LONG")
    field(NELM, "10")
    field(SCAN, "I/O Intr")
    field(TSE,  "-2")
}
```

When the IOC runs, I am setting the TPRO field to 20 which turns on verbose tracing. In another terminal window I am running a command line `camonitor` on the variable.
At one point I am stopping (^D) the IOC, so that the last updates that are traced are the same shown last in the client.

**Case 1:** TSE = -2 (server timestamps)
IOC console:
```
OPC:DDA:wfint32: (server time 2021-05-13 15:03:45.642339200) read incomingData (Good)  array of OpcUa_Int32[10] into epicsInt32[10] --- remaining queue 0/3
OPC:DDA:wfint32: read (state=up, reason=incomingData, status=0, setVal=y) -> 10 array elements read
Element [ROOT] set data (incomingData) for record OPC:DDA:wfint32 (queue use 1/3)
Registering record OPC:DDA:wfint32 for processing (incomingData)
cbLow: dbProcess of 'OPC:DDA:wfint32'
OPC:DDA:wfint32: (server time 2021-05-13 15:03:45.887103900) read incomingData (Good)  array of OpcUa_Int32[10] into epicsInt32[10] --- remaining queue 0/3
OPC:DDA:wfint32: read (state=up, reason=incomingData, status=0, setVal=y) -> 10 array elements read
Element [ROOT] set data (incomingData) for record OPC:DDA:wfint32 (queue use 1/3)
Registering record OPC:DDA:wfint32 for processing (incomingData)
cbLow: dbProcess of 'OPC:DDA:wfint32'
OPC:DDA:wfint32: (server time 2021-05-13 15:03:46.133725900) read incomingData (Good)  array of OpcUa_Int32[10] into epicsInt32[10] --- remaining queue 0/3
OPC:DDA:wfint32: read (state=up, reason=incomingData, status=0, setVal=y) -> 10 array elements read
Element [ROOT] set data (incomingData) for record OPC:DDA:wfint32 (queue use 1/3)
Registering record OPC:DDA:wfint32 for processing (incomingData)
cbLow: dbProcess of 'OPC:DDA:wfint32'
OPC:DDA:wfint32: (server time 2021-05-13 15:03:46.379856500) read incomingData (Good)  array of OpcUa_Int32[10] into epicsInt32[10] --- remaining queue 0/3
OPC:DDA:wfint32: read (state=up, reason=incomingData, status=0, setVal=y) -> 10 array elements read
Element [ROOT] set data (incomingData) for record OPC:DDA:wfint32 (queue use 1/3)
Registering record OPC:DDA:wfint32 for processing (incomingData)
cbLow: dbProcess of 'OPC:DDA:wfint32'
OPC:DDA:wfint32: (server time 2021-05-13 15:03:46.646170500) read incomingData (Good)  array of OpcUa_Int32[10] into epicsInt32[10] --- remaining queue 0/3
OPC:DDA:wfint32: read (state=up, reason=incomingData, status=0, setVal=y) -> 10 array elements read
Element [ROOT] set data (incomingData) for record OPC:DDA:wfint32 (queue use 1/3)
Registering record OPC:DDA:wfint32 for processing (incomingData)
cbLow: dbProcess of 'OPC:DDA:wfint32'
OPC:DDA:wfint32: (server time 2021-05-13 15:03:46.891793700) read incomingData (Good)  array of OpcUa_Int32[10] into epicsInt32[10] --- remaining queue 0/3
OPC:DDA:wfint32: read (state=up, reason=incomingData, status=0, setVal=y) -> 10 array elements read
```
Client output:
```
OPC:DDA:wfint32 2021-05-13 15:03:45.642339 10 32771 32771 32771 32771 32771 32771 32771 32771 32771 32771  
OPC:DDA:wfint32 2021-05-13 15:03:45.887104 10 32774 32774 32774 32774 32774 32774 32774 32774 32774 32774  
OPC:DDA:wfint32 2021-05-13 15:03:46.133726 10 32777 32777 32777 32777 32777 32777 32777 32777 32777 32777  
OPC:DDA:wfint32 2021-05-13 15:03:46.379857 10 32780 32780 32780 32780 32780 32780 32780 32780 32780 32780  
OPC:DDA:wfint32 2021-05-13 15:03:46.646171 10 32783 32783 32783 32783 32783 32783 32783 32783 32783 32783  
OPC:DDA:wfint32 2021-05-13 15:03:46.891794 10 32786 32786 32786 32786 32786 32786 32786 32786 32786 32786  
OPC:DDA:wfint32 2021-05-13 15:03:47.336611 *** disconnected
```
The timestamps match.

**Case 2:** TSE not set (IOC timestamps)
IOC console:
```
OPC:DDA:wfint32: (server time 2021-05-13 15:11:06.894068600) read incomingData (Good)  array of OpcUa_Int32[10] into epicsInt32[10] --- remaining queue 0/3
OPC:DDA:wfint32: read (state=up, reason=incomingData, status=0, setVal=y) -> 10 array elements read
Element [ROOT] set data (incomingData) for record OPC:DDA:wfint32 (queue use 1/3)
Registering record OPC:DDA:wfint32 for processing (incomingData)
cbLow: dbProcess of 'OPC:DDA:wfint32'
OPC:DDA:wfint32: (server time 2021-05-13 15:11:07.140373100) read incomingData (Good)  array of OpcUa_Int32[10] into epicsInt32[10] --- remaining queue 0/3
OPC:DDA:wfint32: read (state=up, reason=incomingData, status=0, setVal=y) -> 10 array elements read
[... 9 lines cut ...]
OPC:DDA:wfint32: read (state=up, reason=incomingData, status=0, setVal=y) -> 10 array elements read
Element [ROOT] set data (incomingData) for record OPC:DDA:wfint32 (queue use 1/3)
Registering record OPC:DDA:wfint32 for processing (incomingData)
cbLow: dbProcess of 'OPC:DDA:wfint32'
OPC:DDA:wfint32: (server time 2021-05-13 15:11:07.887073800) read incomingData (Good)  array of OpcUa_Int32[10] into epicsInt32[10] --- remaining queue 0/3
OPC:DDA:wfint32: read (state=up, reason=incomingData, status=0, setVal=y) -> 10 array elements read
[... cut ...]
OPC:DDA:wfint32 2021-05-13 15:11:07.691541 10 38116 38116 38116 38116 38116 38116 38116 38116 38116 38116  
OPC:DDA:wfint32 2021-05-13 15:11:08.092399 10 38119 38119 38119 38119 38119 38119 38119 38119 38119 38119  
OPC:DDA:wfint32 2021-05-13 15:11:08.293606 10 38122 38122 38122 38122 38122 38122 38122 38122 38122 38122  
OPC:DDA:wfint32 2021-05-13 15:11:08.596918 *** disconnected
```
The IOC timestamps are unrelated, different, always delayed by the time needed for the OPC UA communication (modulo publishing interval) and the IOC record processing.

I think the IOC is handling timestamps correctly.

ralphlange, 2021-05-13:
There is one thing that might explain what you're seeing:

OPC UA always sends two timestamps with every data update: a **server** timestamp and a **source** timestamp.
(Architecture-wise, the OPC UA server sits on top of the device it provides access to. Depending on how the communication between these two is organized, the time that a variable is changed in the source - e.g., read from an IO module - might be different from the time the server updates its representation of that value.)

The IOC database link can specify which of the two stamps to use for a `TSE = -2` setting: `timestamp=server` (default) picks the server timestamp, `timestamp=source` picks the other one. Maybe your LabVIEW client looks at the source timestamp while your IOC uses the server timestamp or vice versa?

ralphlange, 2021-05-13:
Another thing that could happen, depending on the load:

The Channel Access server on the EPICS IOC treats arrays differently from scalar values when updates are queued for sending to the client. Scalar data in the queue has a complete set of timestamp and data, while arrays have the timestamp in the queue and a reference (pointer) to the record which is used when the network buffer is being filled.

That means: if the CA client cannot read the array data fast enough (because of IOC, network or client bandwidth limitation) and the IOC updates the waveform record before the former update has been sent to the client, the CA client will get an update with the old time stamp (from the queue entry) but the new array data (from the referenced record).
That is an unfortunate but known (mis)behaviour of the CA server in the IOC. The OPC UA support can't do anything about it.

You can try to change your `camonitor` client to add the "array" server-side filter on the channel, which has the side-effect of making copies of the array data before putting the update into the queue - that might create the double-buffering you need.

JTGWilson, 2021-05-13:
> There is one thing that might explain what you're seeing:
>
> OPC UA always sends two timestamps with every data update: a **server** timestamp and a **source** timestamp.
> (Architecture-wise, the OPC [...]

This was my problem. I'd not realised that there were the two options and assumed that the **server** timestamp was what I needed. Using the UaExpert client showed the clear difference in these two timestamps (and that **source** matched what I was expecting).

I can now see the correct timestamp in the IOC. 

Thanks Ralph, sorry for flagging my own ignorance as a bug.

closed 2021-05-13 by JTGWilson

ralphlange, 2021-05-13:
Welcome.

I am happy to see that it was the simple reason.
With your 7500 array elements I was fearing it would be the CA misbehaviour, which is much harder to work around.