epics-modules/opcua #114: Make command fails on Windows 10: *** No rule to make target 'RCS/../', needed by '../'.  Stop.

realjinge, 2021-06-21:
Make command fails on Windows 10: 
"Installing CFG file ../../cfg/CONFIG_OPCUA"
"Installing dbd file ../../dbd/opcuaItemRecord.dbd"
mkdir ../../dbd
make[2]: *** No rule to make target 'RCS/../', needed by '../'.  Stop.
make[2]: Leaving directory 'D:/epmcs/epics/support/opcua-old-release-0.5/devOpcuaSup/O.windows-x64'
D:\epmcs\epics\epics-base/configure/RULES_ARCHS:58: recipe for target 'install.windows-x64' failed
make[1]: *** [install.windows-x64] Error 2
make[1]: Leaving directory 'D:/epmcs/epics/support/opcua-old-release-0.5/devOpcuaSup'
D:\epmcs\epics\epics-base/configure/RULES_DIRS:84: recipe for target 'devOpcuaSup.install' failed
make: *** [devOpcuaSup.install] Error 2

-   OPCUA Support: 0.8.0/0.5.2/...
-   Platform: Windows 10
-   EPICS Base: 3.16.2
-   Client library: 1.7.2.493
-   Server: S7-1500

ralphlange, 2021-06-21:
Windows builds are supported since commit 67d8ce0a - which will be part of the soon-to-be-published 0.9 release.

Older releases of the Support do not build or work on Windows.

realjinge, 2021-06-21:
Thanks for your help. I'll try again after new version is released.

closed 2021-06-21 by realjinge

ralphlange, 2021-06-21:
Sure. Shouldn't be long: it's overdue.

ralphlange, 2021-07-05:
Ping! 0.9 is out.