epics-modules/opcua #127: STAT/SEVR are changed to NO_ALARM after setting the variable although connection is down.

jalalmostafa, 2021-09-22:
STAT and SEVR fields are changed from `COMM COMM INVALID` to `NO_ALARM` although there is no connection to OPC UA server.

**To Reproduce**
Steps to reproduce the behavior:
1.  Run a disconnected EPCS server
2.  Monitor a record STAT and SEVR fields
3.  While disconnected, use `caput` to write to the record. 

**Expected behavior**
STAT/SEVR should always show an error if the connection is not working.

**Setup (please complete the following information if applicable):**
-   OPCUA Support: v0.9.3
-   Platform: Ubuntu 21.04
-   EPICS Base: 7.0.6.1
-   Client library: UA SDK 1.7
-   Server: None

**Additional context**
Add any other context about the problem here.

ralphlange, 2021-09-22:
That is for an output record?

jalalmostafa, 2021-09-22:
yes for an `ao`. Specifically:

```
record("ao", "8110d90a-851c-4cf1-b9e6-7d238021d3ab")
{
    field(OUT, "@OPC ns=2;s=0.0_0 monitor=n")
    field(DTYP, "OPCUA")
    field(DESC, "Timestamp 1 Timestamp")
    field(EGU, "None")
    field(MDEL, "-1")
    field(ADEL, "-1")
}
```

ralphlange, 2022-03-05:
Sorry for the long delay.
The issue was indeed introduced by a recent reorganization of the device support that avoids calling into the driver for unconnected items. It applies to both write and read actions. (I.e., when explicitly processing unmonitored read records.)
I have implemented a fix and will merge after the tests have finished.

commit 1b1fb22f (2022-03-05): Always set record to COMM/INVALID when item is disconnected

- Avoids records going into NO_ALARM when processed while
  connection is down.
  (fixes #127)
closed 2022-03-05 by ralphlange