epics-modules/opcua #128: Automatic reconnection function

xuehuawang, 2021-12-21:
**Describe the bug**

After the running epis ioc is disconnected from the device plc1500, reconnect, the epis ioc can automatically establish a connection with the device, but the data will not be refreshed at this time (opcua-v0-5-2 is used).

**To Reproduce**
Steps to reproduce the behavior:
1.  Run epics ioc to establish a connection with a plc1500, at this time the data on the plc changes, the pv value monitored by camonitor will be refreshed；
2.  Disconnect the plc network cable, wait a while, and then plug in the network cable；
3.  The epics ioc changes from the connected state to the disconnected state, and then automatically connects to the plc, but at this time the pv value monitored by camonitor will not be refreshed with the refresh of the plc data.

**Expected behavior**
After epics ioc is disconnected from plc1500, there is an automatic reconnection function, and the data can be refreshed normally after reconnection.

**Screenshots**

[image]

[image]

**Setup (please complete the following information if applicable):**
-   OPCUA Support: \[v0.5.2\]
-   Platform: \[ubuntu18.04\]
-   EPICS Base: \[3.16.2\]
-   Client library: \[UA SDK 1.7.3\]
-   Server: \[S7-1500\]

**Additional context**
Add any other context about the problem here.

ralphlange, 2021-12-21:
The error messages, especially "createMonitoredItems failed with status BadNothingToDo" seems to suggest that your setup does not create monitored items for the variables that you are connecting to.
Please show the definition of a record that behaves this way (from the database file).

Probably unrelated: version 0.5.2 is more than two years old and a lot has changed since then. Please use a more current version to benefit from bug fixes and new features,

xuehuawang, 2021-12-21:
record(ai, "test-OT:caiji") {
    field(DTYP, "OPCUA")
    field(INP, "@FAST ns=3;s=\"DB1\".\"test2\"")
    field(SCAN, "I/O Intr")
}

This error is because I created two, one FAST and one SLOW, and I only used FAST, so the problem of automatic reconnection without refreshing has nothing to do with this error.

> The error messages, especially "createMonitoredItems failed with status BadNothingToDo" seems to suggest that your setup does not create monitored items for the variables that you are connecting to. [...]

ralphlange, 2021-12-21:
Ok, that explains the error message.

However... "fast" being in lowercase letters, the double quotes inside the link not escaped... are you sure this is the definition from the database file? IMHO, the link should look more like
```
field(INP, "@FAST ns=3;s=\"DB1\".\"test2\"")
```
(See also the example database for the S7-1500 or the cheat cheat, as v0.5.2 doesn't have example code.)

To check the state of connections and items, you can use the debugging calls from the IOC shell.
On v0.5.2, you have `opcuaShowSession` and `opcuaShowSubscription`, which take the name (e.g. `FAST`) as the first argument and a verbosity level (small positive integer) as second.
On current versions, there is `opcuaShow`, which takes a wildcard pattern as first and the verbosity as second argument. It works on session, subscription and record names.

xuehuawang, 2021-12-22:
Hello, yesterday’s FAST was because the uppercase automatically changed to lowercase when I copied it...

Today I used the method you suggested, debugged it, and found the problem, but I don’t know how to modify it.

When "connection status changed from connectionErrorApiReconnect to connected", the data can be refreshed normally; when "connection status changed from connectionErrorApiReconnect to NewSessionCreated", the data cannot be refreshed. As shown below:

Attached are the two source code files I located. Thank you very much if you can help me see how to modify them.

王雪华   Xuehua Wang
软件测试工程师
Software Test Engineer
合肥中科离子医学技术装备有限公司
Hefei CAS Ion Medical and Technical Devices Co., Ltd
+86-15385186965
安徽省合肥市高新区方兴大道与柏堰湾路交口
Intersection of Fangxing Avenue and Baiyanwan road in high tech Zone, Hefei, Anhui, China
 
From: Ralph Lange
Date: 2021-12-21 21:35
To: epics-modules/opcua
CC: xuehuawang; Author
Subject: Re: [epics-modules/opcua] Automatic reconnection function (Issue #128)
Ok, that explains the error message.
However... "fast" being in lowercase letters, the double quotes inside the link not escaped... are you sure this is the definition from the database file? IMHO, the link should look more like
field(INP, ***@***.*** ns=3;s=\"DB1\".\"test2\"")

(See also the example database for the S7-1500 or the cheat cheat, as v0.5.2 doesn't have example code.)
To check the state of connections and items, you can use the debugging calls from the IOC shell.
On v0.5.2, you have opcuaShowSession and opcuaShowSubscription, which take the name (e.g. FAST) as the first argument and a verbosity level (small positive integer) as second.
On current versions, there is opcuaShow, which takes a wildcard pattern as first and the verbosity as second argument. It works on session, subscription and record names.

ralphlange, 2021-12-22:
Ok, I see your mail program modifies mail content a lot...
Also, there are no attachments on this GitHub ticket.

In a procedure that is part of the OPC UA standard, in case of a lost connection, the server keeps the resources related to that connection for a configurable time (default: 20 minutes). If the client reconnects within that time, the server will reuse the resources.
On small and medium S7-1500s, we have seen really bad performance while the server is keeping such an "open" connection. It is important to exit the IOC orderly (typing "exit" or using ^D) so that the connection is properly shut down. Using ^C or killing the IOC with the INT or KILL signal will leave the connection broken. (Blocked resources can be freed on the PLC by waiting 20 minutes or by restarting it.)

`ConnectionWarningWatchdogTimeout` means that the client (IOC) finds the server (S7-1500) being unresponsive when doing the periodic ping every few seconds. This usually happens when the server is busy.
`ConnectionErrorApiReconnect` means that the client has lost communication with the server.
`NewSessionCreated` is an intermediate state that means the server was not able to reuse the resources and created a new session. (So that the client has to redo some of the configuration.)
`Connected` means the connection is up and working.

I have seen different paths between those.

- If the PLC starts after the IOC started:
  ErrorApiReconnect -> Connected.

- If you pull the network cable for a short time then put it back:
  Connected -> ErrorApiReconnect -> Connected. (Server was able to reuse the resources of the broken connection.)
 
- If you pull the network for a long time then put it pack, or if you restart the PLC:
  Connected -> ErrorApiReconnect -> NewSessionCreated -> Connected. (Server did a fresh session.)
 
- If the PLC is getting overloaded for a short time:
  Connected -> WarningWatchdogTimeout -> Connected. (The connection was never down.)
 
- If the PLC is getting overloaded for a longer time:
  Connected -> WarningWatchdogTimeout -> ErrorApiReconnect -> Connected. (Client disconnected then reconnected.) 

- If the PLC is getting overloaded for a longer time and drops the connection and its resources:
  Connected -> WarningWatchdogTimeout -> ErrorApiReconnect -> NewSessionCreated -> Connected. (Client got disconnected, then reconnected and the server signaled that the session info was not reused.) 

Because of a few issues with different servers, the connection procedure on the IOC side was completely redone in releases 0.7 and 0.8. Please check if the current version still shows the behavior that you see.

ralphlange, 2022-03-20:
No more reply. Let's assume the problem has been fixed. Closing; please reopen if necessary.

closed 2022-03-20 by ralphlange