epics-modules/opcua #129: when running ioc, it reports "by thread'_main_' failed in ../O.Common/opcuaItemRecord.h line 156."

xuehuawang, 2021-12-21:
**Describe the bug**

As shown in the figure below, when running ioc, it reports "by thread'_main_' failed in ../O.Common/opcuaItemRecord.h line 156."

[image]

**Setup (please complete the following information if applicable):**
-   OPCUA Support: \[v0.9.3\]
-   Platform: \[Ubuntu18.04\]
-   EPICS Base: \[3.16.2\]
-   Client library: \[UA SDK 1.7.3\]
-   Server: \[S7-1500\]

ralphlange, 2021-12-21:
This error appears when the dbd file that the IOC loads (with the ASCII definition of the EPICS database) does not match the support library that it links against. (The opcuaItemRecord changed recently, which would explain that the number of fields does not match when using an older DBD file.)
Please properly recompile the application that creates your IOC binary and the matching dbd. (Do a "make distclean" before building for building from scratch.)

Please note also that EPICS Base 3.16.2 is an unsupported release of EPICS Base. You should use the latest release of EPICS 7 instead.
Also, version 1.7.3 is not the latest release of the UA SDK. I have no reports of any issues with 1.7.3, though, so this should be fine.

ralphlange, 2022-03-20:
No more reply. The issue is most probably a configuration or build problem. Closing as 'invalid'.

closed 2022-03-20 by ralphlange