epics-modules/opcua #136: Windows error when build the module

juanca-rf, 2023-03-31:
Hi, i am a newbie on EPICS, more installing modules. 
I'm trying to build this module in EPICS7 on Windows10 pro but i get this error:

ralphlange, 2023-03-31:
What environment are you using to build on Windows?
What is your configuration?
(See the [Windows how-to](https://docs.epics-controls.org/projects/how-tos/en/latest/getting-started/installation-windows.html) for what I mean.)
Which version of the United Automation C++ SDK are you building against?

The library mentioned in your errors is supposed to be built at that place - which makes your error look like a configuration problem.

juanca-rf, 2023-03-31:
I use the native windows command line. I built EPICS with the first choice of the link. 
What do you mean with the version of UA C++? is another prerequisite?

ralphlange, 2023-04-01:
I will try to reproduce this.
The SDK you installed is the evaluation bundle, I guess?

What configuration have you done in the OPC UA Device support module? (Please share your changes to config files or any config that you added.)

juanca-rf, 2023-04-01:
You guess right, it is the evaluation bundle.

I have not done any special configuration more than set the paths to EPICS and modules:
The configure/RELEASE file is:

```
# Variables and paths to dependent modules:
MODULES = C:\base-R7.0.4.1\modules
#MYMODULE = $(MODULES)/my-module

# Google Test / Google Mock
#GTEST = /path/to/gtest/module

# EPICS_BASE should appear last so earlier modules can override stuff:
EPICS_BASE = C:\base-R7.0.4.1

# Set RULES here if you want to use build rules from somewhere
# other than EPICS_BASE:
#RULES = $(MODULES)/build-rules

# These allow developers to override the RELEASE variable settings
# without having to modify the configure/RELEASE file itself.
-include $(TOP)/../RELEASE.local
-include $(TOP)/../configure/RELEASE.local
-include $(TOP)/configure/RELEASE.local
```
I let commented the path to gtest because i read if i do not define it the build will be clean but without tests.
Also i attach a .txt with the output of the make command.

[MakeErrorBuildingOPCUA.txt](https://github.com/epics-modules/opcua/files/11129889/MakeErrorBuildingOPCUA.txt)

ralphlange, 2023-04-01:
The other thing that needs to be configured is the location of your SDK (evaluation bundle).

Check the README at https://github.com/epics-modules/opcua/tree/master/devOpcuaSup/UaSdk
You will need to create the `CONFIG_SITE.local` file with the SDK location and the other settings shown. (The settings in that README should work for the evaluation bundle.)

juanca-rf, 2023-04-01:
I get the same error and quite the same output  in the make command. I attached another file with the "new" output. Just 21 and 22 lines are diferent.
```
make (e=2): El sistema no puede encontrar el archivo especificado.
make[2]: *** [C:\base-R7.0.4.1/configure/RULES_BUILD:249: devOpcua.obj] Error 2
make[2]: Leaving directory 'C:/base-R7.0.4.1/modules/opcua-master/devOpcuaSup/O.windows-x64'
make[1]: *** [C:\base-R7.0.4.1/configure/RULES_ARCHS:58: install.windows-x64] Error 2
make[1]: Leaving directory 'C:/base-R7.0.4.1/modules/opcua-master/devOpcuaSup'
make: *** [C:\base-R7.0.4.1/configure/RULES_DIRS:85: devOpcuaSup.install] Error 2
```
Also i show you a screenshoot of the enviroment variables of my windows host.
[MakeErrorBuildingOPCUA_2.txt](https://github.com/epics-modules/opcua/files/11130595/MakeErrorBuildingOPCUA_2.txt)

ralphlange, 2024-09-03:
Your error logs show that the shell does not find the "cl" executable, i.e., the VisualStudio compiler.
You need to run the VS setup batch script before doing VS compilations. The EPICS build systems needs "cl" and the other tools of the compiler toolchain to be in the PATH.

As of 73279f3, Windows builds have been added to the CI. These can serve as a source of information on how to build on Windows.

I'm closing this ticket.