epics-modules/opcua #137: Accessing OPCUA nodes on Siemens S7-1200 using signal names

konrad-gajewski, 2023-03-31:
I've been trying to access OPCUA nodes using the identifierString in INP/OUT record fields (s=...) but always got "BadNodeIdUnknown" error. I've tried different syntax (with and without escaped quotes as shown in "EPICS Support for OPC UA - Cheat Sheet.pdf" and including or omitting the first part of the name: SSA (ns=4,i=1).
I wonder if using identifierString  is supported in S7-1200 and I just used the wrong syntax. Accessing the data exposed to the OPC UA server on S7-1200 using identifierIndex works fine so it's not a big problem although I would prefer to use the names instead of numbers to identify the data.

In the PLC the exported data looks like below:
[image]

In the UAExpert the same data looks like this:
[image]

ralphlange, 2023-03-31:
I have never tried it with an S7-1200...

When using UAExpert, the correct string identifier appears in the node details window. That declaration uses the same notation as the link data in the EPICS record:
`ns=<number>; s="<name>"`
Remember that you have to escape embedded quote and space characters properly inside the link data (Siemens uses quotes in names).
(See examples in the [Cheat Sheet](https://github.com/epics-modules/opcua/blob/master/documentation/EPICS%20Support%20for%20OPC%20UA%20-%20Cheat%20Sheet.pdf).)

konrad-gajewski, 2023-03-31:
Yes, I know. I have tried all the combinations below (longin record), SSA1 is a subscription name set in opcuaSubscription()):

    field(INP, "@SSA1 ns=4;s=Counter")
    field(INP, "@SSA1 ns=4;s=\"Counter\"")
    field(INP, "@SSA1 ns=4;s=\"SSA\".\"Counter\"")
    field(INP, "@SSA1 ns=4;s=\"SSA\".Counter")

and for the opcuaItem record:

    field(INP, "@SSA1 ns=4;s=ModbusBuffer")
    field(INP, "@SSA1 ns=4;s=\"ModbusBuffer\"")
    field(INP, "@SSA1 ns=4;s=\"SSA\".\"ModbusBuffer\"")
    field(INP, "@SSA1 ns=4;s=\"SSA\".ModbusBuffer")

ralphlange, 2023-03-31:
Can you share what the "Counter" looks like in the details window of UAExpert?

ralphlange, 2023-04-01:
Thanks, that made it clear.

The EPICS records need to be configured with the NodeId.

The names that you are trying to use are the "BrowseName" or "DisplayName" attributes of a node. They form a hierarchical graph that is used when browsing the server - that's what is shown in the "Address Space" view.

However, the hierarchical BrowseName is not unique - that's why it can't be used to identify a node.
Here's the key sentence from the OPC UA spec:
> Unlike [NodeIds](https://reference.opcfoundation.org/search/13?t=NodeIds), the [BrowseName](https://reference.opcfoundation.org/search/13?t=BrowseName) cannot be used to unambiguously identify a [Node](https://reference.opcfoundation.org/search/13?t=Node). Different [Nodes](https://reference.opcfoundation.org/search/13?t=Nodes) may have the same [BrowseName](https://reference.opcfoundation.org/search/13?t=BrowseName).

(Source: https://reference.opcfoundation.org/Core/Part3/v104/docs/5.2.4)

I am reluctant to allow using BrowseNames in the EPICS links because of that restriction.

Instead, please consider switching your S7-1200 to use string type NodeIds. This seems possible, although not trivial. (https://support.industry.siemens.com/cs/document/109793221/how-do-you-change-the-node-id-identifier-type-of-the-nodes-in-the-s7-1200-opc-ua-server-from-numeric-to-string-?dti=0&lc=en-US)

konrad-gajewski, 2023-04-01:
Thanks Ralph for a really good explanation. I will try to set-up the string NodeIds in my PLC and will report here on how it worked out.

ralphlange, 2023-04-01:
Great. I'll be interested to know how it went.

konrad-gajewski, 2023-04-03:
After changing the identifierType to String (using Method 1 described in https://support.industry.siemens.com/cs/document/109793221/how-do-you-change-the-node-id-identifier-type-of-the-nodes-in-the-s7-1200-opc-ua-server-from-numeric-to-string-?dti=0&lc=en-US I can access the PVs using ns=...;s=... scheme. 
It works fine when using the "flat" names, for example     field(INP, "@SSA1 ns=4;s=\"ModbusBuffer\".\"BufferINOut\".\"ADCdata\".\"Idrain\"")
I have tried to use opcuaItem record and access the elements of the structures using element=... identifier but it didn't work, It looks like the information about the variable structure is missing - I get some warnings when connecting to the OPC server from UaExpert.
[image]

If there is any advantage of using opcuaItem record instead of "flat" addressing I will dig in further otherwise I will leave it as it is.

ralphlange, 2023-05-24:
Did you get the user-defined structures working eventually?

Siemens says (and we agree) that using structures has a big performance benefit. If you have many data items (thousands and more), updating a single structured node is 10-100x more performant on the PLC side compared to getting the updates using a separate node for each element of the structure.

My colleagues are running into the same issue right now (they switched to string node IDs, but structures don't work).