epics-modules/opcua #138: Error Building on Ubuntu 18: Can't open perl script "/makeFromList.pl": No such file or directory

juanca-rf, 2023-04-19:
Hi again and accept my apologies for being a newbie. 

I'm trying again on Linux, with the SDK installed (with gcc 6.3 as a requisite of the SDK) I get the following errors:
```
Can't open perl script "/makeFromList.pl": No such file or directory
/home/juanca/EPICS/epics-base/cfg/RULES_GTEST:42: recipe for target 'NamespaceMapTest.make' failed
make[3]: *** [NamespaceMapTest.make] Error 2
make[3]: Leaving directory '/home/juanca/EPICS/epics-base/modules/opcua/unitTestApp/src/O.linux-x86_64'
/home/juanca/EPICS/epics-base/configure/RULES_ARCHS:58: recipe for target 'install.linux-x86_64' failed 
make[2]: *** [install.linux-x86_64] Error 2
make[2]: Leaving directory '/home/juanca/EPICS/epics-base/modules/opcua/unitTestApp/src'
/home/juanca/EPICS/epics-base/configure/RULES_DIRS:85: recipe for target 'src.install' failed
make[1]: *** [src.install] Error 2
make[1]: Leaving directory '/home/juanca/EPICS/epics-base/modules/opcua/unitTestApp'
/home/juanca/EPICS/epics-base/configure/RULES_DIRS:85: recipe for target 'unitTestApp.install' failed
make: *** [unitTestApp.install] Error 2

```
I let you all the output of make after a "make clean" and also the reduced one after another make.

[MakeAfterClean.txt](https://github.com/epics-modules/opcua/files/11276083/MakeAfterClean.txt)
[MakeAfterMake.txt](https://github.com/epics-modules/opcua/files/11276084/MakeAfterMake.txt)

ralphlange, 2023-04-27:
First thing I notice:
You've installed the opcua module _inside_ of EPICS Base. That might be unrelated to the issue you see, but still: The 'module' directory inside EPICS Base is intended for the modules that are part of EPICS Base, not for Device Support modules.

juanca-rf, 2023-04-29:
That worked! Thank you so much.

closed 2023-04-29 by juanca-rf