epics-modules/opcua #141: Writing to write-only variables fails

dirk-zimoch, 2023-05-26:
We have an OPC UA server with some write-only variables.
Writing to them causes the error message: `unsupported conversion for outgoing data`

```
opcuaShow CRIO-9046:BO-1 1
item ns=2;s=Data.Node1 record=CRIO-9046:BO-1 state=up status=BadNotReadable dataDirty=n context=SUB1@OPS1
sampling=0(-1) qsize=1(1) cqsize=3 discard=old timestamp=server bini=read output=y monitor=y registered=-(n)
leaf=[ROOT] record(bo)=CRIO-9046:BO-1 type=OpcUa_Null timestamp=server bini=read monitor=y
```

The  data type is Null. This type is deduced from `incomingData`, which of course had not been set during startup, because the variable is not readable. See also: `status=BadNotReadable.`

ralphlange, 2023-05-26:
Interesting.

I have never seen a write-only variable before. That will need some effort...

Things I see in your configuration:
- `bini=read` cannot work if the node is not readable
- `monitor=y` neither

How should we get the right definition of the OPC UA data type, if the node is not readable? Does a client like UA Expert still show the type?

dirk-zimoch, 2023-05-26:
>  `bini=read` cannot work if the node is not readable
    `monitor=y` neither

I think those are simply the defaults. The `OUT` link reads:
`   field (OUT,  "@SUB1 ns=2;s=$(SetParam)") `

This is what uaexpert sees the type as "Boolean":
[image]

ralphlange, 2023-05-26:
I know these are the defaults - just saying that these defaults will not work for write-only variables, as they configure the client to read at connection-up and set up a monitor.

Ok.
So UaExpert can read the type without having read access to the data. I need to find out how I can get to that information.

dirk-zimoch, 2023-05-26:
The only things uaexpert logs for this is:
```
Attribute Plugin | OPC UA Server | Read attributes of node 'NS2|String|Data.Node2' succeeded [ret = Good].
Reference Plugin | OPC UA Server | Browse succeeded.
```

ralphlange, 2023-05-26:
"Read attributes" - sounds good.

ralphlange, 2023-05-26:
Found it. I need to use the browse (introspection) interface of the server to do that. That gets me access to all the attributes of a node without reading the value.

Up to now I was just reading the value, which also gets me the data type.

dirk-zimoch, 2023-05-26:
Maybe do that only if initialRead fails because of BadNotReadable?

ralphlange, 2023-05-26:
Yes. Otherwise I would have to always add a lot of OPC UA traffic just to cover that rare case.