epics-modules/opcua #146: Empty options should be ignored

dirk-zimoch, 2023-09-19:
`opcuaSession` and `opcuaSubscription` accept a list of options, but empty list elements cause the warning:
`option '' must follow 'key=value' format - ignored`

Empty options are useful in script snippets with macros like:
`opcuaSession $(OPC_SESS) opc.tcp://$(OPC_HOST):$(OPC_PORT=4840) sec-mode=$(OPC_SEC=None):$(OPC_SESS_OPT=)`

When `OPC_SESS_OPT` is given, the `:` before is required, but when `OPC_SESS_OPT` is empty (the default) the terminating `:` causes a warning.

ralphlange, 2023-09-19:
Reasonable.

dirk-zimoch, 2023-09-19:
PS: I would have preferred to have the options in an `iocshArgArgv` instead of using a `iocshArgString` with separators.

ralphlange, 2023-09-19:
That was obviously to keep it consistent with the single argument to the `opcuaSession` and `opcuaSubscription` calls.

But we could add the other as an optional format, couldn't we?

ralphlange, 2023-09-19:
Oh, you mean in the `opcuaSession` call?

dirk-zimoch, 2023-09-19:
Yes, in the `opcuaSession` and `opcuaSubscription` calls. Their last argument is:
`static const iocshArg opcuaSubscriptionArg3 = {"[options]", iocshArgString};`
But for arbitrary length lists of arguments (at the end) we already have `iocshArgArgv`.

ralphlange, 2023-09-21:
I was not aware of `iocshArgArgv`. I'll have a look. Thanks for the pointer!

dirk-zimoch, 2023-09-21:
It is a structure. `arg[n.].aval.ac` contains the number of remaining arguments (from position `n` on) and `arg[n].aval.av` is an array of `char*` with the remaining args as strings, followed by a `NULL`. 

There is only one strangeness with `iocshArgArgv`. If it is the first (i.e. only argument), element 0 `arg[0].aval.av[0]` is not the first argument but the command itself. The first argument is in `arg[0].aval.av[1]`.

dirk-zimoch, 2023-09-21:
I can do the necessary changes and make a PR.

dirk-zimoch, 2023-09-21:
PR #147

dirk-zimoch, 2023-09-21:
> There is only one strangeness with iocshArgArgv. If it is the first (i.e. only argument), element 0 arg[0].aval.av[0] is not the first argument but the command itself. The first argument is in arg[0].aval.av[1].

I remembered it wrong: `arg[n].aval.av[0]` *always* contains a (string) copy of the previous argument `arg[n-1]`  (or the command if `n==0`).

ralphlange, 2024-09-05:
Closed by merge of PR #147.

closed 2024-09-05 by ralphlange