epics-modules/opcua #148: Array type matches are too strict

dirk-zimoch, 2023-09-21:
Currently, the EPICS type of the array record must match the OPC-UA data type quite precisely. It is not possible to load arrays into a record of a different (even larger) type, e.g. UInt16 into LONG.

This is particularly annoying with arrays of Boolean, as there is no matching EPICS type.

It is no problem to load any integer type into a longin record, even Boolen and unsigned types. The same should be possible for arrays.

ralphlange, 2023-09-21:
This was an intentional design decision, based on performance concerns.

If sizes match, arrays can be copied using `memcpy()`. If sizes don't match, we need to loop and copy element by element.

ralphlange, 2023-09-21:
For OPC UA Boolean, I think UCHAR arrays work.