epics-modules/opcua #151: Windows: INSTALL deploy mode does not work

ralphlange, 2023-11-21:
Building an application also needs the DLL stub library in `$(OPCUA)/lib/$(T_A)`, not just the DLL in `$(OPCUA)/bin/$(T_A)`.
(Careful: the stub is named differently between MSVC and MinGW.)

The application, on the other hand, should only install a copy of the DLL for distribution. The stub is not needed for running an application, just for building it.

splashmak, 2024-12-13:
We are trying to build it on windows (minGW) and we get :
**make[2]: *** No rule to make target '../opcua.dll', needed by '../../lib/windows-x64-mingw/opcua.dll'.  Stop.**
You think it could be linked ? how can we choose between MSVC & MinGW ? our EPICS_HOST_ARCH is configured in a environnement variable

ralphlange, 2024-12-13:
What are you trying to build?

splashmak, 2024-12-13:
opcuaBinaryDist

ralphlange, 2024-12-13:
Which doesn't exist for Windows.
(I don't even have a Windows license of the SDK.)

ralphlange, 2024-12-13:
For Windows builds, you can use the UASDK evaluation bundle. (I'm testing with those.)

Long-term, I would recommend using the open62541 client. Free, open, and works on Windows.

splashmak, 2024-12-13:
In fact we don't know where to begin with windows ( linux is working but we want to test it on windows too).
So you confirm that we have to download **Source code (zip)**  and not BDIST ?

ralphlange, 2025-08-06:
Closing as wontfix.