epics-modules/opcua #153: Compiling for other EPICS base versions

gilesknap, 2024-01-09:
Hi Ralf,

This is not really a feature request - more of a question, I have been trying to contact your email it keeps getting delivery errors.

I'm trying to make use of your OPC  UA support module inside of our epics-containers framework. The base image that I'm using for all epics containers is here:-

https://github.com/epics-containers/epics-base

It is based on Ubuntu 22.04 and epics base 7.0.7. I've tried to use your binary package for this release of Ubuntu. But I get these link errors because my base version does not match.
```
/usr/bin/ld: warning: libdbCore.so.3.15.9, needed by /epics/ioc-ts99i-ps-01/ioc/opcuaIocApp/libopcua.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libCom.so.3.15.9, needed by /epics/ioc-ts99i-ps-01/ioc/opcuaIocApp/libopcua.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: /epics/ioc-ts99i-ps-01/ioc/opcuaIocApp/libopcua.so: undefined reference to `epicsMutex::epicsMutex()'
/usr/bin/ld: /epics/ioc-ts99i-ps-01/ioc/opcuaIocApp/libopcua.so: undefined reference to `epicsTime::epicsTime()'
/usr/bin/ld: /epics/ioc-ts99i-ps-01/ioc/opcuaIocApp/libopcua.so: undefined reference to `epicsTime::strftime(char*, unsigned long, char const*) const'
/usr/bin/ld: /epics/ioc-ts99i-ps-01/ioc/opcuaIocApp/libopcua.so: undefined reference to `epicsTime::operator epicsTimeStamp() const'
```
I've also tried compiling from source using your excellent instructions but open62541-1.3 fails due to deprecation warnings since the version of openSSL does not like the API calls open62541 is using.

I'd like to be able to use this withing my framework without having to make a special base image for this one support module.

How would you recommend I proceed?

Many thanks,
giles

closed 2024-01-09 by gilesknap

ralphlange, 2024-01-09:
The SDK identifies as "Open62541 Client API v1.2.2" - are you sure you're compiling and using 1.3?

If you can live without security for the moment, you could also try explicitly configuring Open62541 to build without security. That should avoid any of the calls into the OpenSSL library that eventually cause the crash.

The Open62541 project focuses on the server. The client does mainly exist as a proof of concept and to support tests. I'm afraid this shows in the priority that client-side bugs get.

gilesknap, 2024-01-09:
Thanks Ralph, I had not noticed that. I was pretty sure I had 1.3 but will take another look.