epics-modules/opcua #168: compilation issue  BDIST_opcua-0.10.0_Base-7.0.8.1_rhel9.tar.gz

splashmak, 2024-12-11:
I follow the "HOW_TO.md" help file. Everything seems ok but when I launch both demo Apps I get 
"../../bin/linux-x86_64/opcuaIoc: error while loading shared libraries: libdbRecStd.so.3.23.1: cannot open shared object file: No such file or directory"

**To Reproduce**
I get a Centos 9, 
 - yum update
 -  yum -y install gcc-c++ perl make **readline ( and not libreadline )**
 - yum -y install openssl-devel libxml2-devel

Working on directory **/home/admin/epics-opcua**

- epics base 7.0.8.1 is ok ( I can launch softIOC and co.). Please not that I removed my other EPICS_BASE and EPICS_HOST_ARCH environnement variable used for my S7nodave IOC.
- make -j2 -C opcuaBinaryDist  no error but no ./bin folder
- make -j2 -C opcuaExampleIoc /home/admin/epics-opcua/opcuaExampleIoc/bin/linux-x86_64/opcuaIoc present but when launching it with the st.cmd ( iocS7-1500 AND iocUaDemoServer) I get : 
  ../../bin/linux-x86_64/opcuaIoc: error while loading shared libraries: libdbRecStd.so.3.23.1: cannot open shared object file: No such file or directory

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Setup (please complete the following information if applicable):**
-   OPCUA Support:  0.10.0 
-   Platform: Centos 9
-   EPICS Base: 7.0.8
-   Client library: /
-   Server: Will try to work with S7-1500 to replace S7-nodave

splashmak, 2024-12-11:
Thank you for your HELP!!!

splashmak, 2024-12-11:
OK it works for examples. But is it normal that I still don't have bin folder for opcuaBinaryDist ?

splashmak, 2024-12-11:
Ok we can close... I had read base 7.0.8
FAIL! sorry for spamming

ralphlange, 2024-12-12:
Glad you found it!

Yes, as EPICS Base uses the full version number as SONAME for its libraries, you need to compile against the exact matching version of Base.
(At least, it fails at compile time and doesn't produce core dumping IOC binaries...)

ralphlange, 2024-12-12:
`opcuaBinaryDist` only creates a library, so there's no binary to install and the bin directory is not created. All good and expected.