epics-modules/opcua #171: Compilation error on WSL2 Ubuntu.

beldish, 2025-02-06:
/usr/bin/g++  -D_GNU_SOURCE -D_DEFAULT_SOURCE          -D_X86_64_ -DUNIX  -Dlinux      -O3 -g   -Wall    -DUSE_TYPED_RSET -DEPICS_NO_CALLBACK -DHAS_SECURITY -DHAS_XMLPARSER -std=c++11  -mtune=generic     -m64  -fPIC -I. -I../O.Common -I. -I. -I.. -I../../devOpcuaSup -I../../devOpcuaSup/open62541 -I../../include/compiler/gcc -I../../include/os/Linux -I../../include    -I/home/mbeldis/work/epics/base/include/compiler/gcc -I/home/mbeldis/work/epics/base/include/os/Linux -I/home/mbeldis/work/epics/base/include    -I../../devOpcuaSup/open62541 -I/usr/include/libxml2 -I/home/mbeldis/work/epics/open62541/include    -c ../../devOpcuaSup/open62541/Session.cpp
In file included from /usr/local/include/open62541/config.h:226,
                 from /home/mbeldis/work/epics/open62541/include/open62541/types.h:20,
                 from /home/mbeldis/work/epics/open62541/include/open62541/client.h:22,
                 from ../../devOpcuaSup/open62541/SessionOpen62541.h:25,
                 from ../../devOpcuaSup/open62541/Session.cpp:14:
/usr/lib/gcc/x86_64-linux-gnu/11/include/stdatomic.h:40:9: error: ‘_Atomic’ does not name a type
   40 | typedef _Atomic _Bool atomic_bool;
      |         ^~~~~~~
/usr/lib/gcc/x86_64-linux-gnu/11/include/stdatomic.h:41:9: error: ‘_Atomic’ does not name a type
   41 | typedef _Atomic char atomic_char;
      |         ^~~~~~~
/usr/lib/gcc/x86_64-linux-gnu/11/include/stdatomic.h:42:9: error: ‘_Atomic’ does not name a type
   42 | typedef _Atomic signed char atomic_schar;
      |         ^~~~~~~

$ uname -a
Linux J0242 5.15.90.1-microsoft-standard-WSL2 #1 SMP Fri Jan 27 02:56:13 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux

$ /usr/bin/g++ --version
g++ (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0

Searched in issues in hope somebody have already reported but can not find one.

Thanks

ralphlange, 2025-02-06:
This is more an open62541 problem (as `stdatomic.h` is included from `open62541/config.h`)...

Which version of open62541 is that? Compiled using which settings?

ralphlange, 2025-02-06:
Also 
```
In file included from /usr/local/include/open62541/config.h:226,
from /home/mbeldis/work/epics/open62541/include/open62541/types.h:20,
from /home/mbeldis/work/epics/open62541/include/open62541/client.h:22,
```
Seems you are using a mixture of include files from your local build and a build that was installed in open62541's default install location (`/usr/local`). Doesn't look healthy.

beldish, 2025-02-07:
Thanks you for prompt reply.

I cloned open62541 from git repository:
```
git clone https://github.com/open62541/open62541.git
mkdir build
cd build/
cmake -DBUILD_SHARED_LIBS=ON  ..
make
```
After this ```config.h``` file is generated in ```./build/src_generated/open62541/config.h``` and obviously is not accessible by opcua even if I set in configure/CONFIG_SITE.local
```
OPEN62541=/home/mbeldis/work/epics/open62541
```
That is why I did in open62541  directory
```
sudo make install 
```  
And this is the reason why config.h is included from /usr/local and rest from my home directory. 

Anyway I updated configure/CONFIG_SITE.local to
```
open62541=/usr/local
```

and get same errors.

```
/usr/bin/g++  -D_GNU_SOURCE -D_DEFAULT_SOURCE           -D_X86_64_  -DUNIX  -Dlinux      -O3 -g   -Wall     -DUSE_TYPED_RSET -DEPICS_NO_CALLBACK -DHAS_SECURITY -DHAS_XMLPARSER -std=c++11 -std=c++11  -mtune=generic      -m64  -fPIC -I. -I../O.Common -I. -I. -I.. -I../../devOpcuaSup -I../../devOpcuaSup/open62541 -I../../include/compiler/gcc -I../../include/os/Linux -I../../include    -I/home/mbeldis/work/epics/base/include/compiler/gcc -I/home/mbeldis/work/epics/base/include/os/Linux -I/home/mbeldis/work/epics/base/include     -I../../devOpcuaSup/open62541 -I/usr/include/libxml2 -I/usr/local/include    -c ../../devOpcuaSup/open62541/Session.cpp
In file included from /usr/local/include/open62541/config.h:226,
                 from /usr/local/include/open62541/types.h:20,
                 from /usr/local/include/open62541/client.h:22,
                 from ../../devOpcuaSup/open62541/SessionOpen62541.h:25,
                 from ../../devOpcuaSup/open62541/Session.cpp:14:
/usr/lib/gcc/x86_64-linux-gnu/9/include/stdatomic.h:40:9: error: ‘_Atomic’ does not name a type
   40 | typedef _Atomic _Bool atomic_bool;
      |         ^~~~~~~
/usr/lib/gcc/x86_64-linux-gnu/9/include/stdatomic.h:41:9: error: ‘_Atomic’ does not name a type
   41 | typedef _Atomic char atomic_char;
      |         ^~~~~~~
/usr/lib/gcc/x86_64-linux-gnu/9/include/stdatomic.h:42:9: error: ‘_Atomic’ does not name a type
   42 | typedef _Atomic signed char atomic_schar;
      |         ^~~~~~~
/usr/lib/gcc/x86_64-linux-gnu/9/include/stdatomic.h:43:9: error: ‘_Atomic’ does not name a type
   43 | typedef _Atomic unsigned char atomic_uchar;
      |         ^~~~~~~
/usr/lib/gcc/x86_64-linux-gnu/9/include/stdatomic.h:44:9: error: ‘_Atomic’ does not name a type

```

Now all includes are from /usr/local. These were same anyway.

I have read this has something to do with g++ not recognizing Atomic unless -std=c++11 is set.
And it is set (even twice :) )
But am lost here as in another project I am using atomic with C++ but different standard c++17.

PS: Thanks for your time. Appreciated.

ralphlange, 2025-02-07:
Let me cite the first sentence of the README in the open62541 subdirectory of this module, under prerequisites:

> [Open62541](https://www.open62541.org/) SDK / C library (v1.2 and v1.3 series have been tested; their system packages should also work. open62541 v1.4 changed the header file structure and does not work yet.)

ralphlange, 2025-02-07:
If you get open62541 v1.4 working, we'd be very interested in a PR!

Otherwise, please use an open62541 release of the 1.2 or 1.3 series.

closed 2025-02-07 by ralphlange

beldish, 2025-02-07:
Thank you very much. 
If time permits I may look into 1.4.
Thanks again. All the best.