epics-modules/opcua #174: Can't compile opcuaExampleIoc with Debian 12

marcoutcho, 2025-02-20:
**Describe the bug**
Since we decide to move for Centos with Debian I Redo de compilation : 
- opcuaBinaryDist : Is compiled
- opcuaExampleIoc :  not working

**To Reproduce**
Steps to reproduce the behavior:

I put the RELEASE.local file with EPICS_BASE and OPCUA

**Expected behavior**
bin folder with IOC application compiled

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Setup (please complete the following information if applicable):**
-   OPCUA Support:   v0.10         
-   Platform:   Debian 12                             
-   EPICS Base:  7.0.8.1

**Additional context**

control@EPICS:~/epics/modules/opcua/opcuaExampleIoc$ make
[ ... ]
make[3]: Nothing to be done for 'install'.
make[3]: Leaving directory '/home/control/epics/modules/opcua/opcuaExampleIoc/DeviceDbApp/UaDemoServerDb/O.linux-x86_64'
make[2]: Leaving directory '/home/control/epics/modules/opcua/opcuaExampleIoc/DeviceDbApp/UaDemoServerDb'
make[1]: Leaving directory '/home/control/epics/modules/opcua/opcuaExampleIoc/DeviceDbApp'
make -C ./opcuaIocApp install
make[1]: Entering directory '/home/control/epics/modules/opcua/opcuaExampleIoc/opcuaIocApp'
make -C O.linux-x86_64 -f ../Makefile TOP=../.. \
    T_A=linux-x86_64 install
make[2]: Entering directory '/home/control/epics/modules/opcua/opcuaExampleIoc/opcuaIocApp/O.linux-x86_64'

/usr/bin/g++ -o opcuaIoc  -L/home/control/epics/modules/opcua/opcuaExampleIoc/lib/linux-x86_64 -L/home/control/epics/base-7.0.8.1/lib/linux-x86_64 -L/home/control/epics/modules/opcua/opcuaBinaryDist/lib/linux-x86_64 -Wl,-rpath,/home/control/epics/modules/opcua/opcuaExampleIoc/lib/linux-x86_64 -Wl,-rpath,/home/control/epics/base-7.0.8.1/lib/linux-x86_64 -Wl,-
rpath,/home/control/epics/modules/opcua/opcuaBinaryDist/lib/linux-x86_64          -rdynamic -m64         opcuaIoc_registerRecordDeviceDriver.o opcuaIocMain.o    -lopcua -ldbRecStd -ldbCore -lca -lCom  **-lxml2
/usr/bin/ld: cannot find -lxml2: No such file or directory**
collect2: error: ld returned 1 exit status

make[2]: *** [/home/control/epics/base-7.0.8.1/configure/RULES_BUILD:231: opcuaIoc] Error 1
make[2]: Leaving directory '/home/control/epics/modules/opcua/opcuaExampleIoc/opcuaIocApp/O.linux-x86_64'
make[1]: *** [/home/control/epics/base-7.0.8.1/configure/RULES_ARCHS:58: install.linux-x86_64] Error 2
make[1]: Leaving directory '/home/control/epics/modules/opcua/opcuaExampleIoc/opcuaIocApp'
make: *** [/home/control/epics/base-7.0.8.1/configure/RULES_DIRS:85: opcuaIocApp.install] Error 2

Since it was ok with my Centos I think I related to a package missing in Debian but can't find which one.

Thank you for your help

marcoutcho, 2025-02-20:
Ok
libxml2-devel needed was libxml2-dev on debian

closed 2025-02-20 by marcoutcho

ralphlange, 2025-02-20:
Thanks for this (short-lived) ticket!
I see that the dependency on `libxml2` is visible in the fine print, but not prominent enough. I will make it a bit more obvious in the READMEs.