epics-modules/opcua #175: Connecting with credentials

marcoutcho, 2025-02-21:
**Describe the bug**
We compiled our own IOC ( using makeBaseApp), IOC is running. We already made some tests without security and we now want to add it gradually. So first step is to use "none" with login/password - without encryption. 

Here are lines in st.cmd  : 
opcuaOptions OPCIBA sec-mode=None
#opcuaOptions OPCIBA sec-policy=default
opcuaOptions OPCIBA sec-id="./credential"

I put a credential file in the same place as the st.cmd one with
BadLogin;BadPassword 

**Expected behavior**
I'm expecting to see "invalid credentials" like message but it seems everything is going well.
on UaExpert without credentials it's not working.

**Screenshots**
[image]
[image]

**Setup (please complete the following information if applicable):**
-   OPCUA Support: 0.10
-   Platform: Debian 12
-   EPICS Base:  7.0.8.1
-   Server: S7-1500

Furthermore I didn't see how to format the credential file.

Again thank you very much for your AMAZING support

ralphlange, 2025-02-21:
**Be aware that the binary distribution will not provide OPC UA Security on Debian 12 - the openssl v3 libraries don't work with version v1.7.4 of the UA SDK that I have access to.**

For OPC UA Security you need to either use an older Debian or a newer UA SDK (buy or use evaluation bundle) - or you switch to using the free open62541 SDK that works fine on older and newer systems.

Assuming you switch to the evaluation bundle (that's what my Deb-12 test machine has)...

For the credentials file format, see https://github.com/epics-modules/opcua/blob/master/Using-Security.md#identity-client-authentication

Your IOC is in the TOP directory (it does a `cd "/home/control/epics/ioc/opc"`), so your identity file in the IOC's startup directory must have the path from TOP: `opcuaOptions OPCIBA sec-id=iocBoot/iocikon/credential`

Doing things wrong will always get you a warning message, like:

```
############################################################################
## EPICS R7.0.8.2-DEV
## Rev. 2025-01-04T16:52+0100
## Rev. Date build date/time: 
############################################################################
OPC UA Client Device Support 0.11.0 (-); using Unified Automation C++ Client SDK v1.8.4-632
iocRun: All initialization complete
OPC UA: Autoconnecting sessions
OPC UA session OPC1: credentials file iocBoot/iocS7-1500/credential does not contain settings for Username token (user + pass) or Certificate token (cert + key [+ pass])
```

To use authentication (even the simple user/pass version on an insecure connection), you need to accept the server certificate. (Credentials will always be encrypted, even on an insecure connection - else they would be sent as cleartext.)

Check the certificate status with the command `opcuaShowSecurity <session>`.
```
epics> opcuaShowSecurity OPC1
Session OPC1    (discovery at opc.tcp://localhost:4840)
  Server name: SIMATIC.S7-1500.OPC-UAServer:PLC_1
  Server URI:  urn:SIMATIC.S7-1500.OPC-UAServer:PLC_1
  Server URL:  opc.tcp://10.130.1.122:4840    (using opc.tcp://localhost:4840)
  Requested security mode: None    policy: None
  Identity: Username token 'BlaBlaBla' (credentials from iocBoot/iocS7-1500/credential)
  ----- Endpoint 0 ----- Level:   0 --------------------------------------------------
    Security Mode: None              Policy: None
    URL: opc.tcp://10.130.1.122:4840    (using opc.tcp://localhost:4840)
    Server Certificate: PLC-1/OPCUA-1-1 (Siemens) serial 353A759D (thumb 4D7DB2E98681445FE7E85C627F482874A72E647D) self-signed - trusted
  ----- Endpoint 1 ----- Level:  55 --------------------------------------------------
    Security Mode: Sign              Policy: Basic128Rsa15
    URL: opc.tcp://10.130.1.122:4840    (using opc.tcp://localhost:4840)
    Server Certificate: PLC-1/OPCUA-1-1 (Siemens) serial 353A759D (thumb 4D7DB2E98681445FE7E85C627F482874A72E647D) self-signed - trusted
  ----- Endpoint 2 ----- Level: 105 --------------------------------------------------
    Security Mode: SignAndEncrypt    Policy: Basic128Rsa15
    URL: opc.tcp://10.130.1.122:4840    (using opc.tcp://localhost:4840)
    Server Certificate: PLC-1/OPCUA-1-1 (Siemens) serial 353A759D (thumb 4D7DB2E98681445FE7E85C627F482874A72E647D) self-signed - trusted
  ----- Endpoint 3 ----- Level:  60 --------------------------------------------------
    Security Mode: Sign              Policy: Basic256
    URL: opc.tcp://10.130.1.122:4840    (using opc.tcp://localhost:4840)
    Server Certificate: PLC-1/OPCUA-1-1 (Siemens) serial 353A759D (thumb 4D7DB2E98681445FE7E85C627F482874A72E647D) self-signed - trusted
  ----- Endpoint 4 ----- Level: 110 --------------------------------------------------
    Security Mode: SignAndEncrypt    Policy: Basic256
    URL: opc.tcp://10.130.1.122:4840    (using opc.tcp://localhost:4840)
    Server Certificate: PLC-1/OPCUA-1-1 (Siemens) serial 353A759D (thumb 4D7DB2E98681445FE7E85C627F482874A72E647D) self-signed - trusted
```
The string `trusted` is the sign that things should connect now.

Set the debug level on the session to see more details of the connection process. With `debug=10` you see something like
```
############################################################################
## EPICS R7.0.8.2-DEV
## Rev. 2025-01-04T16:52+0100
## Rev. Date build date/time: 
############################################################################
OPC UA Client Device Support 0.11.0 (-); using Unified Automation C++ Client SDK v1.8.4-632
iocRun: All initialization complete
OPC UA: Autoconnecting sessions
Session OPC1: (disconnect) already disconnected (Disconnected)
Session OPC1: (setupClientSecurityInfo) setting up PKI provider
Session OPC1: (setupClientSecurityInfo) no client certificate configured
Session OPC1: (setupIdentity) setting Username token (BlaBlaBla/*****)
Session OPC1: (setupSecurity) no security configured 
Session OPC1: connection status changed from Disconnected to ConnectionErrorApiReconnect
Session OPC1: connect service succeeded
## Start any sequence programs
#seq sncopcuaIoc,"user=ralph"
```
which shows (not very obvious, but...) that the user/pass authentication is used and rejected (-> ConnectionErrorApiReconnect is the state where the UA SDK client tries reconnecting).

Once the credentials are correct, things look better:
```
iocInit
Starting iocInit
############################################################################
## EPICS R7.0.8.2-DEV
## Rev. 2025-01-04T16:52+0100
## Rev. Date build date/time: 
############################################################################
OPC UA Client Device Support 0.11.0 (-); using Unified Automation C++ Client SDK v1.8.4-632
iocRun: All initialization complete
OPC UA: Autoconnecting sessions
OPC UA session OPC1: connected as 'ralph' (username token) (sec-mode: None; sec-policy: None)
OPC UA session OPC1: WARNING - this session uses *** NO SECURITY ***
## Start any sequence programs
#seq sncopcuaIoc,"user=ralph"
epics> 
```
Yay!

ralphlange, 2025-02-21:
See also #176 - the error message should be improved.

The connection failure looks the same no matter what goes wrong - I don't see a way to distinguish between an untrusted or missing certificate and wrong credentials.

marcoutcho, 2025-02-21:
For you, what's the best configuration for an Industrial production system?

ralphlange, 2025-02-24:
Discussion continued by mail. Closing the ticket.