epics-modules/opcua #177: (requestRead) beginRead service failed with status BadTooManyOperations

marcoutcho, 2025-02-25:
**Describe the bug**
Got this message when trying to retrieve 5979 bool ( = epics bi) :  
_**(requestRead) beginRead service failed with status BadTooManyOperations**_

**To Reproduce**
Steps to reproduce the behavior:
- All the pv are in one .db file
- Tried opcuaSubscription with 100ms AND 1000ms
- Tried with and without opcuaOptions PLC nodes-max=10.000
- I get a INVALID DRIVER COMM (0) 0 when I do a PVGET so IOC not working

**Expected behavior**
Since srs says : 
- Signals: many. (>50.000)
- Read and write access (including structured data).
- Update period: medium. (200ms)

We want to achieve : 

- 6790 : 200ms
- 2439: 1000ms
- 497: 500ms

9726 pv which seems to be ok regarding SRS

**Setup :**

- OPCUA Support: v0.10
- Platform: Debian 12
- EPICS Base: 7.0.8.1

ralphlange, 2025-02-25:
Looks like your server does not support operations with too many nodes?

Note that the S7 server has a hard-coded limit of 1000 noder per call, so you have to set the `batch-nodes` parameter.

See https://github.com/epics-modules/opcua/wiki/Siemens-S7-1500-PLC

marcoutcho, 2025-02-25:
We are using a Siemens 1518-f with Intouch intouch wonderware using OPC and it's working so PLC/OPC server should not be the issue.

ralphlange, 2025-02-25:
Set nodes-max to 1000, then. Again: The Siemens OPC UA server has a hardcoded limit of 1000 nodes _per call_.

Have you read the Wiki page?

> Note: This is not a limitation on the total number of nodes. (See "System Limits" below.) This setting will just split up larger requests into multiple requests of that size.

marcoutcho, 2025-02-27:
Working we can close