epics-modules/opcua #182: Dependency on build libaries while building an IOC

eddybl, 2025-04-01:
**Describe the bug**

Building an IOC against the OPC-UA module requires the presence of `libxml2-dev` module (probably the same with the ssl libs), although this should only be needed to build the OPC-UA module itself

**To Reproduce**
Steps to reproduce the behavior:
1.  Setup server without `libxml2-dev`, but installed OPC-UA module
2.  Try to build an IOC with OPC-UA support
3.  See build error

**Expected behavior**
It should build without `libxml2-dev`

**Build Log**
```
[...]
make -C O.linux-x86_64 -f ../Makefile TOP=../../.. \
    T_A=linux-x86_64 install
make[3]: Entering directory '/home/ws/hi0724/iocs/magnets/magnetsApp/src/O.linux-x86_64'
/usr/bin/g++  -D_GNU_SOURCE -D_DEFAULT_SOURCE          -D_X86_64_ -DUNIX  -Dlinux      -O3 -g   -Wall      -mtune=generic     -m64  -I. -I../O.Common -I. -I. -I.. -I../../../include/compiler/gcc -I../../../include/os/Linux -I../../../include      -I/opt/epics/modules/opcua-0.10.0/include -I/opt/epics/base-7.0.9/include/compiler/gcc -I/opt/epics/base-7.0.9/include/os/Linux -I/opt/epics/base-7.0.9/include        -MM -MF magnetsMain.d  ../magnetsMain.cpp
Creating dbd file magnets.dbd
perl -CSD /opt/epics/base-7.0.9/bin/linux-x86_64/dbdExpand.pl   -I. -I.. -I../O.Common -I../../../dbd -I/opt/epics/modules/opcua-0.10.0/dbd -I/opt/epics/base-7.0.9/dbd -o magnets.dbd base.dbd opcua.dbd
perl -CSD /opt/epics/base-7.0.9/bin/linux-x86_64/registerRecordDeviceDriver.pl   -I. -I.. -I../O.Common -I../../../dbd -I/opt/epics/modules/opcua-0.10.0/dbd -I/opt/epics/base-7.0.9/dbd    -o magnets_registerRecordDeviceDriver.cpp \
            ../O.Common/magnets.dbd magnets_registerRecordDeviceDriver /home/ws/hi0724/iocs/magnets
/usr/bin/g++  -D_GNU_SOURCE -D_DEFAULT_SOURCE          -D_X86_64_ -DUNIX  -Dlinux      -O3 -g   -Wall      -mtune=generic     -m64  -I. -I../O.Common -I. -I. -I.. -I../../../include/compiler/gcc -I../../../include/os/Linux -I../../../include      -I/opt/epics/modules/opcua-0.10.0/include -I/opt/epics/base-7.0.9/include/compiler/gcc -I/opt/epics/base-7.0.9/include/os/Linux -I/opt/epics/base-7.0.9/include        -MM -MF magnets_registerRecordDeviceDriver.d  magnets_registerRecordDeviceDriver.cpp
Installing created dbd file ../../../dbd/magnets.dbd
mkdir ../../../dbd
/usr/bin/g++  -D_GNU_SOURCE -D_DEFAULT_SOURCE          -D_X86_64_ -DUNIX  -Dlinux      -O3 -g   -Wall      -mtune=generic     -m64  -I. -I../O.Common -I. -I. -I.. -I../../../include/compiler/gcc -I../../../include/os/Linux -I../../../include      -I/opt/epics/modules/opcua-0.10.0/include -I/opt/epics/base-7.0.9/include/compiler/gcc -I/opt/epics/base-7.0.9/include/os/Linux -I/opt/epics/base-7.0.9/include        -c magnets_registerRecordDeviceDriver.cpp
/usr/bin/g++  -D_GNU_SOURCE -D_DEFAULT_SOURCE          -D_X86_64_ -DUNIX  -Dlinux      -O3 -g   -Wall      -mtune=generic     -m64  -I. -I../O.Common -I. -I. -I.. -I../../../include/compiler/gcc -I../../../include/os/Linux -I../../../include      -I/opt/epics/modules/opcua-0.10.0/include -I/opt/epics/base-7.0.9/include/compiler/gcc -I/opt/epics/base-7.0.9/include/os/Linux -I/opt/epics/base-7.0.9/include        -c ../magnetsMain.cpp
/usr/bin/g++ -o magnets  -L/home/ws/hi0724/iocs/magnets/lib/linux-x86_64 -L/opt/epics/base-7.0.9/lib/linux-x86_64 -L/opt/epics/modules/opcua-0.10.0/lib/linux-x86_64 -Wl,-rpath,/home/ws/hi0724/iocs/magnets/lib/linux-x86_64 -Wl,-rpath,/opt/epics/base-7.0.9/lib/linux-x86_64 -Wl,-rpath,/opt/epics/modules/opcua-0.10.0/lib/linux-x86_64          -rdynamic -m64         magnets_registerRecordDeviceDriver.o magnetsMain.o    -lopcua -ldbRecStd -ldbCore -lca -lCom  -lcrypto -lxml2 
/usr/bin/ld: cannot find -lxml2: No such file or directory
collect2: error: ld returned 1 exit status
make[3]: *** [/opt/epics/base-7.0.9/configure/RULES_BUILD:232: magnets] Error 1
make[3]: Leaving directory '/home/ws/hi0724/iocs/magnets/magnetsApp/src/O.linux-x86_64'
make[2]: *** [/opt/epics/base-7.0.9/configure/RULES_ARCHS:58: install.linux-x86_64] Error 2
make[2]: Leaving directory '/home/ws/hi0724/iocs/magnets/magnetsApp/src'
make[1]: *** [/opt/epics/base-7.0.9/configure/RULES_DIRS:85: src.install] Error 2
make[1]: Leaving directory '/home/ws/hi0724/iocs/magnets/magnetsApp'
make: *** [/opt/epics/base-7.0.9/configure/RULES_DIRS:85: magnetsApp.install] Error 2
```

**Setup (please complete the following information if applicable):**
-   OPCUA Support: 0.10.0
-   Platform: Ubuntu 24.04
-   EPICS Base:  7.0.9
-   Client library: open62541

**Additional context**
A "quick" work around is:

- Rename the `modules/opcua-0.10.0/cfg` folder to `cfg.removed` and then the IOC builds

ralphlange, 2025-04-01:
In a setup using shared libraries, the shared libraries for libxml2 and libssl are still needed on the system, as they are dependencies of the Device Support module.
You shouldn't need the `-dev` packages, containing static libraries and header files, though.

Oh, I see, `-lcrypto` and `-lxml2` on the IOC link line require `.so` files, and these are indeed soft links in the `-dev` packages. Maybe these two need to be removed from the linker line?

ralphlange, 2025-04-02:
I'll have a PR for you to try - stay tuned...